bugsounet/MMM-GoogleAssistant #107: News Playback not supported on this Device

joebiwankenobi, 2020-07-27:
Hi Bugsonney
first of all.. Thank you so much, i have tried several Google AI's on my SmartMirror but this one is brilliant as it gives visual feedback and that is exactly what i wanted.
I have pretty much finished the Mirror Now but i noticed this GoogleAssistant does not play the news.
I did use GassistPi before and this was able to play the latest news.
Is this something i have done wrong or is this just not supported yet?

I ask "play the latest news"
the response is "Unable to play the news, not yet supported on this device"

I have added this to my "Google Home App" and it DOES have BBC News assigned to it and did used to work on the last version of Google AI i used.

if not - is there a way to add this?

Many  thanks in advance & Kind Regards

bugsounet, 2020-07-27:
unfortunately this function is not yet available with google Assisant SDK. ([limitation there](https://developers.google.com/assistant/sdk/overview))
Google Assistant SDK is +/- a demo of a real Google Assistant and i port it to MagicMirror

I did not ask myself how to provide this functionality.
currently MMM-News is largely enough for me (RSS feed)

joebiwankenobi, 2020-07-27:
Understood Cedric.

Many thanks for the speedy reply and yes – I already have an RSS News feed on the bottom bar.

Kind Regards

closed 2020-07-28 by bugsounet