bugsounet/MMM-GoogleAssistant #110: Hotwords and wakeup

effelle, 2020-07-30:
First of all, thanks for this almost perfect module, very clean and efficient considering the limitation of Google SDK.

After reading some old issues, I understood you are not using an external `Snowboy` module but in fact your own. 
I then understand why is not possible to freely add a new hotword but would be nice at least to have the classic "ok google" available.
As a background information, consider my whole house have a Google Home device on every room, I even have one nest mini on the bathroom of my master room. 
The only device I need to call with another name is the mirror, so please consider to add at least the official hotword, thanks.

Also, would be nice to have a feature to turn on the screen when the hotword is said. As for now I'm using a IR barrier to detect a person well in front of the screen but I would like to have the screen on every time the assistant is triggered (if possible) like when I sit on the couch, far from the mirror. 

Thanks for your attention.

F.

bugsounet, 2020-07-31:
Hi,

> First of all, thanks for this almost perfect module, very clean and efficient considering the limitation of Google SDK.

thanks, I try to exploit the native responses that google returns

> After reading some old issues, I understood you are not using an external Snowboy module but in fact your own.
I then understand why is not possible to freely add a new hotword but would be nice at least to have the classic "ok google" available.

humm, yes and no ;) because it's orginal snowboy project but i maintened only node version and correct some errors
(before compilation was hell... now it take 2 sec!)
and Unfortunately I can't create new release keyword and snowboy website have not release it. (maybe Google don't want ?)
Kitt.ai website will turn off snowboy in the end of the year
That's why I don't add train mode for personal keyword (personnal keyword is only valid 30 days before retraining...) and use only official keyword
so actually i search a real alternative, simple and configurable (and with Personal hotkey !)

> Also, would be nice to have a feature to turn on the screen when the hotword is said. As for now I'm using a IR barrier to detect a person well in front of the screen but I would like to have the [...]

this feature is already coded.
Let's try MMM-Assistant2Display extension with a lot of sub-modules whose screen and pir manager (turn on screen on presence (pir) and turn on screen on call assistant, keep on when playing video and music, turn off screen with timer)

as you can see I try to do the best and optimize as much as possible for a simple RPI

effelle, 2020-07-31:
Then I guess I need to get used to a different keyword. First Snips.ai was gone, now Snowboy... It's a pity.

I'll have a look on MMM-Assistant2Display, even if I'm not using a RPi but an old notebook, thanks.

> as you can see I try to do the best and optimize as much as possible for a simple RPI

I know the feeling, I code for ESP devices and fitting all the stuff we have in such a tiny space with low processing capacity is a nightmare but... It works! :)

Closing as I get all the information I need, thanks again for your time.

F.

closed 2020-07-31 by effelle