bugsounet/MMM-GoogleAssistant #111: Supprimer MMM-AssistantMk2 ?

JohnB7777, 2020-07-31:
Bonjour,

Je me permets de vous parler en français, car il me semble que c'est votre langue.
Je créé un incident ici car je n'ai pas trouvé un autre moyen de vous contacter.
J'ai depuis plusieurs années le module MMM-AssistantMk2 et MMM-Hotword installés sur mon raspberry-pi (qui n'a d'ailleurs qu'1 Go de RAM, je pense que je devrai le changer...).
J'aurais voulu savoir si je devais supprimer ces deux modules et installer MMM-GoogleAssistant en suivant toutes les étapes ou s'il y avait un moyen de les mettre à jour ?
J'ai aussi remarqué que MMM-Hotword n'existe plus, est-ce que cette fonctionnalité est disponible dans MMM-GoogleAssistant ?

Merci :)

bugsounet, 2020-07-31:
Hi,

MMM-AssistantMk2 and MMM-Hotword is now oudated. I have +/- deprecied and downgraded MMM-AssistantMk2 works only with audio reponse.
For MMM-Hotword (snowboy based), this module is not needed at now because MMM-GoogleAssistant have is own embed and  use my personal snowboy library

So if you want to upgrade it:

* save your AMK2 `credentials.json` and token file (maybe in ./profiles/default.json)
* don't do the google setup setup (because it's work on AMk2) ;)

* place your `credentials.json` inside the MMM-GoogleAssistant directory
* place your `default.json` file inside your MMM-GoogleAssistant directory and rename it to `token.json`
* do the `npm install` step and  configure it

JohnB7777, 2020-07-31:
Merci pour cette réponse très rapide !
D'ailleurs, comme je parle français, vous n'étiez pas obligé de me répondre en anglais (bien que ça pourrait aider d'autres personnes) ;)
Encore merci ! :)

bugsounet, 2020-07-31:
C'est le but ;)

JohnB7777, 2020-07-31:
Hello again,

I follow all the steps and I have several errors when I do "npm start" :  

[23:17:37.001] [LOG]    Starting MagicMirror: v2.10.1
[23:17:37.019] [LOG]    Loading config ...
[23:17:37.031] [LOG]    Loading module helpers ...
[23:17:37.033] [LOG]    No helper found for module: alert.
[23:17:37.035] [LOG]    No helper found for module: clock.
[23:17:37.915] [LOG]    Initializing new module helper ...
[23:17:37.917] [LOG]    Module helper loaded: calendar
[23:17:37.918] [LOG]    No helper found for module: currentweather.
[23:17:37.919] [LOG]    No helper found for module: weatherforecast.
[23:17:37.921] [LOG]    No helper found for module: MMM-AirQuality.
[23:17:38.008] [LOG]    Initializing new module helper ...
[23:17:38.009] [LOG]    Module helper loaded: MMM-network-signal
[23:17:38.035] [LOG]    Initializing new module helper ...
[23:17:38.036] [LOG]    Module helper loaded: MMM-Tools
[23:17:38.098] [LOG]    Initializing new module helper ...
[23:17:38.099] [LOG]    Module helper loaded: MMM-NetworkScanner
[23:17:38.808] [LOG]    Initializing new module helper ...
[23:17:38.809] [LOG]    Module helper loaded: MMM-EARTH
**[23:17:39.147] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: @grpc/grpc-js only works on Node ^8.13.0 || >=10.10.0**
[23:17:39.149] [LOG]    Loading module helpers ...
[23:17:39.150] [LOG]    No helper found for module: alert.
[23:17:39.152] [LOG]    No helper found for module: clock.
[23:17:39.153] [LOG]    Initializing new module helper ...
[23:17:39.154] [LOG]    Module helper loaded: calendar
[23:17:39.155] [LOG]    No helper found for module: currentweather.
[23:17:39.156] [LOG]    No helper found for module: weatherforecast.
[23:17:39.157] [LOG]    No helper found for module: MMM-AirQuality.
[23:17:39.158] [LOG]    Initializing new module helper ...
[23:17:39.159] [LOG]    Module helper loaded: MMM-network-signal
[23:17:39.160] [LOG]    Initializing new module helper ...
[23:17:39.161] [LOG]    Module helper loaded: MMM-Tools
[23:17:39.162] [LOG]    Initializing new module helper ...
[23:17:39.163] [LOG]    Module helper loaded: MMM-NetworkScanner
[23:17:39.164] [LOG]    Initializing new module helper ...
[23:17:39.165] [LOG]    Module helper loaded: MMM-EARTH
[23:17:39.173] [ERROR]  App threw an error during load
**[23:17:39.174] [ERROR]  Error: @grpc/grpc-js only works on Node ^8.13.0 || >=10.10.0**
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@grpc/grpc-js/build/src/index.js:49:11)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@grpc/grpc-js/build/src/index.js:118:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js:5:14)
[23:17:39.177] [LOG]    Whoops! There was an uncaught exception...
**[23:17:39.178] [ERROR]  Error: @grpc/grpc-js only works on Node ^8.13.0 || >=10.10.0**
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@grpc/grpc-js/build/src/index.js:49:11)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@grpc/grpc-js/build/src/index.js:118:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js:5:14)
[23:17:39.185] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[23:17:39.186] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
**[23:17:39.194] [ERROR]  (node:7149) ExperimentalWarning: The http2 module is an experimental API.**
[23:17:39.481] [LOG]    Launching application.

I try "npm run rebuild" but nothing change.

Thanks for helping !

bugsounet, 2020-07-31:
the problem is there: `[23:17:39.178] [ERROR] Error: @grpc/grpc-js only works on Node ^8.13.0 || >=10.10.0`

first verify node version (should be outdated and less as 10.10.0)
```
$ node -v
v10.22.0
```
v10.22.0 is the lasted

so try this:
```
sudo apt-get update
sudo apt-get upgrade
sudo npm install -g npm
```
1 first command update RPI package list
2 second command order to update all package
3 third command order to update npm command to lasted version (v6.14.7)

after verify if node version is updated (node -v) and verify npm version too (npm -v)

if it's ok. retry the npm run rebuild command inside the GA directory

JohnB7777, 2020-08-01:
Hello,

Thanks for helping but after following your steps, the error still there :

```
npm version
{ magicmirror: '2.10.1',
  npm: '6.14.7',
  ares: '1.15.0',
  brotli: '1.0.7',
  cldr: '35.1',
  http_parser: '2.9.3',
  icu: '64.2',
  modules: '64',
  napi: '6',
  nghttp2: '1.41.0',
  node: '10.22.0',
  openssl: '1.1.1g',
  tz: '2019c',
  unicode: '12.1',
  uv: '1.34.2',
  v8: '6.8.275.32-node.58',
  zlib: '1.2.11' }
```

```
node -v
v10.22.0
```

[22:45:01.943] [LOG]    Starting MagicMirror: v2.10.1
[22:45:01.959] [LOG]    Loading config ...
[22:45:01.971] [LOG]    Loading module helpers ...
[22:45:01.974] [LOG]    No helper found for module: alert.
[22:45:01.975] [LOG]    No helper found for module: clock.
[22:45:02.851] [LOG]    Initializing new module helper ...
[22:45:02.852] [LOG]    Module helper loaded: calendar
[22:45:02.853] [LOG]    No helper found for module: currentweather.
[22:45:02.854] [LOG]    No helper found for module: weatherforecast.
[22:45:02.856] [LOG]    No helper found for module: MMM-AirQuality.
[22:45:02.942] [LOG]    Initializing new module helper ...
[22:45:02.943] [LOG]    Module helper loaded: MMM-network-signal
[22:45:02.970] [LOG]    Initializing new module helper ...
[22:45:02.971] [LOG]    Module helper loaded: MMM-Tools
[22:45:03.032] [LOG]    Initializing new module helper ...
[22:45:03.033] [LOG]    Module helper loaded: MMM-NetworkScanner
[22:45:03.735] [LOG]    Initializing new module helper ...
[22:45:03.736] [LOG]    Module helper loaded: MMM-EARTH
**[22:45:03.999] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: @grpc/grpc-js only works on Node ^8.13.0 || >=10.10.0**
[22:45:04.001] [LOG]    Loading module helpers ...
[22:45:04.002] [LOG]    No helper found for module: alert.
[22:45:04.003] [LOG]    No helper found for module: clock.
[22:45:04.004] [LOG]    Initializing new module helper ...
[22:45:04.005] [LOG]    Module helper loaded: calendar
[22:45:04.006] [LOG]    No helper found for module: currentweather.
[22:45:04.007] [LOG]    No helper found for module: weatherforecast.
[22:45:04.008] [LOG]    No helper found for module: MMM-AirQuality.
[22:45:04.009] [LOG]    Initializing new module helper ...
[22:45:04.010] [LOG]    Module helper loaded: MMM-network-signal
[22:45:04.011] [LOG]    Initializing new module helper ...
[22:45:04.012] [LOG]    Module helper loaded: MMM-Tools
[22:45:04.013] [LOG]    Initializing new module helper ...
[22:45:04.013] [LOG]    Module helper loaded: MMM-NetworkScanner
[22:45:04.014] [LOG]    Initializing new module helper ...
[22:45:04.015] [LOG]    Module helper loaded: MMM-EARTH
[22:45:04.022] [ERROR]  App threw an error during load
**[22:45:04.023] [ERROR]  Error: @grpc/grpc-js only works on Node ^8.13.0 || >=10.10.0**
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@grpc/grpc-js/build/src/index.js:49:11)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@grpc/grpc-js/build/src/index.js:118:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js:5:14)
[22:45:04.026] [LOG]    Whoops! There was an uncaught exception...
**[22:45:04.027] [ERROR]  Error: @grpc/grpc-js only works on Node ^8.13.0 || >=10.10.0**
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@grpc/grpc-js/build/src/index.js:49:11)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@grpc/grpc-js/build/src/index.js:118:3)
    at Module._compile (internal/modules/cjs/loader.js:711:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:722:10)
    at Module.load (internal/modules/cjs/loader.js:620:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:559:12)
    at Function.Module._load (internal/modules/cjs/loader.js:551:3)
    at Module.require (internal/modules/cjs/loader.js:658:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js:5:14)
[22:45:04.033] [LOG]    MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[22:45:04.034] [LOG]    If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
**[22:45:04.042] [ERROR]  (node:30986) ExperimentalWarning: The http2 module is an experimental API.**
[22:45:04.235] [LOG]    Launching application.

What can I do ?

Thans again !

bugsounet, 2020-08-02:
Hi,

GoogleAssistant need this: node >=10.10.0 in the MagicMirror core

I test with v2.10.x version of MagicMirror:
 * node version installed for MM core is v10.2.0 (I can't update it)

the best way is updating MagicMirror to v2.12.0
 * node version installed for MM core is v12.4.0

So try this:

```sh
cd ~/MagicMirror
git pull
npm install
cd ~/MagicMirror/modules/MMM-GoogleAssistant
npm run rebuild
```
Notes : 
That's why you can see in [pre-requirements](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#pre-requirements):
MagicMirror v2.11.0 and more

JohnB7777, 2020-08-02:
Thanks, it's working !
I have just to say, if anyone else read these lines, that I had some problems with others modules (because the node version wasn't the same).
I had to go to these module's files and writte `npm install `for each module which was in error.

Thanks again :)

bugsounet, 2020-08-02:
perfect, now let's try extention if you want :) --> MMM-Assistant2Display (see wiki)

closed 2020-08-02 by bugsounet