bugsounet/MMM-GoogleAssistant #113: Error during Installation

Feedy88, 2020-08-06:
Hey there,

while installing MMM-GoogleAssistant running `npm install` results in an error. Log below:

`pi@MirrorPi:~/MagicMirror/modules/MMM-GoogleAssistant $ npm install

> MMM-GoogleAssistant@ preinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/preinstall.sh

Welcome to MMM-GoogleAssistant

No root checking...
Checking MMM-GoogleAssistant: pi/pi
Checking MagicMirror: pi/pi

Checking OS...
OS Detected: linux-gnueabihf (raspbian 10 armv7l)

Checking all dependencies...
You must install missing dependencies before going further
Missing package: libasound2-dev
Missing package: gcc-7
Missing package: mpg321
Attempt to automatically install the above packages? [Y/n]
Your choice: y
Installing missing package...
Holen:1 http://archive.raspberrypi.org/debian buster InRelease [32,6 kB]
Holen:2 http://raspbian.raspberrypi.org/raspbian buster InRelease [15,0 kB]
Holen:3 https://deb.nodesource.com/node_10.x buster InRelease [4.584 B]
Holen:4 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13,0 MB]
Holen:5 http://archive.raspberrypi.org/debian buster/main armhf Packages [331 kB]
Holen:6 https://deb.nodesource.com/node_10.x buster/main armhf Packages [765 B]
Holen:7 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58,7 kB]
Es wurden 13,4 MB in 9 s geholt (1.576 kB/s).
Paketlisten werden gelesen... Fertig
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  gyp libc-ares2 libjs-inherits libjs-is-typedarray libssl-dev libuv1 libuv1-dev node-abbrev node-ajv node-ansi
  node-ansi-align node-ansi-regex node-ansi-styles node-ansistyles node-aproba node-archy node-are-we-there-yet
  node-asn1 node-assert-plus node-asynckit node-aws-sign2 node-aws4 node-balanced-match node-bcrypt-pbkdf
  node-bluebird node-boxen node-brace-expansion node-builtin-modules node-builtins node-cacache node-call-limit
  node-camelcase node-caseless node-chalk node-chownr node-cli-boxes node-cliui node-clone node-co node-color-convert
  node-color-name node-combined-stream node-concat-map node-concat-stream node-config-chain
  node-console-control-strings node-copy-concurrently node-core-util-is node-cross-spawn node-cyclist node-dashdash
  node-decamelize node-decompress-response node-deep-extend node-defaults node-delayed-stream node-delegates
  node-detect-indent node-detect-newline node-duplexer3 node-duplexify node-ecc-jsbn node-editor node-encoding
  node-end-of-stream node-errno node-escape-string-regexp node-execa node-extend node-extsprintf node-find-up
  node-flush-write-stream node-forever-agent node-form-data node-from2 node-fs-vacuum node-fs-write-stream-atomic
  node-fs.realpath node-gauge node-get-caller-file node-get-stream node-getpass node-glob node-got node-graceful-fs
  node-har-schema node-har-validator node-has-flag node-has-symbol-support-x node-has-to-string-tag-x node-has-unicode
  node-hosted-git-info node-http-signature node-iconv-lite node-iferr node-import-lazy node-imurmurhash node-inflight
  node-inherits node-ini node-invert-kv node-is-builtin-module node-is-npm node-is-object node-is-plain-obj
  node-is-retry-allowed node-is-stream node-is-typedarray node-isarray node-isexe node-isstream node-isurl node-jsbn
  node-json-parse-better-errors node-json-schema node-json-stable-stringify node-json-stringify-safe node-jsonify
  node-jsonparse node-jsonstream node-jsprim node-latest-version node-lazy-property node-lcid node-libnpx
  node-locate-path node-lockfile node-lowercase-keys node-lru-cache node-mem node-mime-types node-mimic-fn
  node-mimic-response node-minimatch node-minimist node-mississippi node-mkdirp node-move-concurrently
  node-mute-stream node-node-uuid node-nopt node-normalize-package-data node-npm-package-arg node-npm-run-path
  node-npmlog node-oauth-sign node-object-assign node-once node-opener node-os-locale node-osenv node-p-cancelable
  node-p-finally node-p-limit node-p-locate node-p-timeout node-package-json node-parallel-transform node-path-exists
  node-path-is-absolute node-path-is-inside node-performance-now node-prepend-http node-process-nextick-args
  node-promise-inflight node-promzard node-proto-list node-prr node-pump node-pumpify node-punycode node-qs node-qw
  node-rc node-read node-read-package-json node-readable-stream node-registry-auth-token node-registry-url
  node-request node-require-directory node-require-main-filename node-resolve-from node-retry node-rimraf
  node-run-queue node-safe-buffer node-semver node-semver-diff node-set-blocking node-sha node-shebang-command
  node-shebang-regex node-signal-exit node-slash node-slide node-sorted-object node-spdx-correct
  node-spdx-expression-parse node-spdx-license-ids node-sshpk node-ssri node-stream-each node-stream-iterate
  node-stream-shift node-string-decoder node-string-width node-strip-ansi node-strip-eof node-strip-json-comments
[... 291 lines cut ...]
fixing sandbox permissions

Installing Sources...
Branch 'prod' folgt nun Remote-Branch 'prod' von 'origin'.
Bereits aktuell.

Installing all npm libraries...
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm ERR! code E404
npm ERR! 404 Not Found - GET https://registry.npmjs.org/@grpc%2fgrpc-js - Not found
npm ERR! 404
npm ERR! 404  '@grpc/grpc-js@^1.1.3' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it (or use the name yourself!)
npm ERR! 404 It was specified as a dependency of '@bugsounet/google-assistant'
npm ERR! 404
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, http url, or git url.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-08-06T09_58_52_629Z-debug.log`

I was not sure if this was an issue, so I went on with setting up GA but failed to run `auth_and_test.js`

Would appreciate some help :)

bugsounet, 2020-08-06:
> npm ERR! 404 '@grpc/grpc-js@^1.1.3' is not in the npm registry.

Weird, it looks like it can't find the library `@grpc/grpc-js`
The 404 error is like the browser (not found page)
However it exists in the [npm registry](https://www.npmjs.com/package/@grpc/grpc-js)

Maybe a temporary error?

radokristof, 2020-08-10:
I have the same problem. Tried installing grpc manually, but still the same error, I don't know why...

bugsounet, 2020-08-10:
i don't use `grpc` library, i use `@grpc/grpc-js`

radokristof, 2020-08-10:
Well, I was too generic. My GoogleAssistant was already installed and it won't start now, because grpc is missing.
`[2020-08-11 00:20:48.042] [ERROR]  Error: Cannot find module 'grpc'

Require stack:

- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js

- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/index.js

- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js

- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js

- /home/pi/MagicMirror/js/app.js

- /home/pi/MagicMirror/js/electron.js

- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js

- 
`

Tried installing grpc-js manually in the GoogleAssistant directory, same problem...

bugsounet, 2020-08-10:
it's means that you don't update properly

please retry with this command in terminal:
`npm run rebuild`

note:
* this command wil detele all node_module and do a clean install
* for next update: try `npm run update` command

Feedy88, 2020-08-11:
I could successfully install today after cleaning a manual installation of snowboy.

So far, I can start my MM but will only see a black screen (no other modules). Log itself looks fine:

```
> magicmirror@2.13.0-develop start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-08-12 01:19:29.365] [LOG]    Starting MagicMirror: v2.13.0-develop
[2020-08-12 01:19:29.372] [LOG]    Loading config ...
[2020-08-12 01:19:29.377] [LOG]    Loading module helpers ...
[2020-08-12 01:19:29.379] [LOG]    No helper found for module: alert.
[2020-08-12 01:19:29.384] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:29.385] [LOG]    Module helper loaded: MMM-WatchDog
[2020-08-12 01:19:29.423] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:29.423] [LOG]    Module helper loaded: updatenotification
[2020-08-12 01:19:29.424] [LOG]    No helper found for module: clock.
[2020-08-12 01:19:29.781] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:29.781] [LOG]    Module helper loaded: calendar
[2020-08-12 01:19:29.782] [LOG]    No helper found for module: MMM-Multimonth.
[2020-08-12 01:19:29.812] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:29.812] [LOG]    Module helper loaded: MMM-MyGarbage
[2020-08-12 01:19:30.260] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:30.261] [LOG]    Module helper loaded: MMM-Spotify
[2020-08-12 01:19:30.262] [LOG]    No helper found for module: MMM-WiFiPassword.
[2020-08-12 01:19:30.262] [LOG]    No helper found for module: MMM-WeatherOrNot.
[2020-08-12 01:19:30.264] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:30.264] [LOG]    Module helper loaded: MMM-Vrr
[2020-08-12 01:19:30.286] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:30.287] [LOG]    Module helper loaded: newsfeed
[2020-08-12 01:19:30.288] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:30.289] [LOG]    Module helper loaded: mmm-systemtemperature
[2020-08-12 01:19:31.172] [LOG]    Initializing new module helper ...
[2020-08-12 01:19:31.173] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-08-12 01:19:31.173] [LOG]    All module helpers loaded.
[2020-08-12 01:19:31.292] [LOG]    Starting server on port 8080 ...
[2020-08-12 01:19:31.301] [LOG]    Server started ...
[2020-08-12 01:19:31.302] [LOG]    Connecting socket for: MMM-WatchDog
[2020-08-12 01:19:31.303] [LOG]    Starting module helper: MMM-WatchDog
[2020-08-12 01:19:31.304] [LOG]    Connecting socket for: updatenotification
[2020-08-12 01:19:31.304] [LOG]    Connecting socket for: calendar
[2020-08-12 01:19:31.305] [LOG]    Starting node helper for: calendar
[2020-08-12 01:19:31.305] [LOG]    Connecting socket for: MMM-MyGarbage
[2020-08-12 01:19:31.306] [LOG]    Starting node_helper for module: MMM-MyGarbage
[2020-08-12 01:19:31.307] [LOG]    Connecting socket for: MMM-Spotify
[2020-08-12 01:19:31.307] [LOG]    Connecting socket for: MMM-Vrr
[2020-08-12 01:19:31.308] [LOG]    Starting module helper: MMM-Vrr
[2020-08-12 01:19:31.308] [LOG]    Connecting socket for: newsfeed
[2020-08-12 01:19:31.309] [LOG]    Starting node helper for: newsfeed
[2020-08-12 01:19:31.309] [LOG]    Connecting socket for: mmm-systemtemperature
[2020-08-12 01:19:31.310] [LOG]    Starting node helper: mmm-systemtemperature
[2020-08-12 01:19:31.310] [LOG]    Connecting socket for: MMM-GoogleAssistant
[2020-08-12 01:19:31.311] [LOG]    Sockets connected & modules started ...
[2020-08-12 01:19:31.407] [LOG]    Launching application.
```

_If you want me to do the troubleshooting somewhere else (other issue post/forum), let me know, I will close this issue in this case._

sdetweil, 2020-08-11:
open the developers window, ctrl-shift-i on the pi keyboard, select the tab labeled console, and scroll up to find any errors, usually red text..

the black screen means that some code in the UI (browser) side of MM died.. 

find the error and then we can work thru that

bugsounet, 2020-08-12:
> magicmirror@2.13.0-develop 

```js
VM138:7 Uncaught SyntaxError: Unexpected token / in JSON at position 203
    at JSON.parse (<anonymous>)
    at XMLHttpRequest.xhr.onreadystatechange (translator.js:22)
```

MM v2.13.0 core have some change for translate file.
So il will correct all translate json code of GA
I will be available in 10 min ;)

bugsounet, 2020-08-12:
Ok, tested in v2.13 core and now it's correct :)
try a simple `git pull` in GA directory for update new translated files

Feedy88, 2020-08-12:
Thanks, the `git pull` did the trick. Needed to adapt my Mic config, working fine now.

General question, what kind of mic and speaker are you guys using? I bought a very cheap one of both but am really not happy with the audio output. Is a separate USB Soundcard reccomended? Happy to get reply via DM, closing this thread.

closed 2020-08-12 by Feedy88