bugsounet/MMM-GoogleAssistant #114: audio open error : Device or resource busy

masoudl, 2020-08-07:
Hi,

First of all thank you for a well documented project and detailed WiKi.
When we start the mirror, I get this error:
[2020-08-06 21:01:12.561] [LOG]    [LPCM16] WARN: arecord: main:828: 
[2020-08-06 21:01:12.563] [LOG]    [LPCM16] WARN: audio open error: Device or resource busy
...
Notes:
- Using Voice Kit 1 on Raspberry 3.b
- The instructions in config.js state that micConfig values should come from auto-installer (see below). But I couldn't find anywhere in the installation process where I would see these values, so I am going w/ arecord and plughw:0.
    micConfig: { // put there configuration generated by auto-installer
      recorder: "arecord",
      device: "plughw:0",
    },

- If I change plughw to another value, I get a different error (see below):
[LPCM16] WARN: ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
arecord: main:828: 
[2020-08-06 21:15:14.061] [LOG]    [LPCM16] WARN: audio open error: No such file or directory
[2020-08-06 21:15:14.069] [LOG]    [SNOWBOY] Retry restarting...
...

Any thoughts on what could be wrong?

Thanks

bugsounet, 2020-08-07:
hi,

thanks for your feedback about the wiki : 

For you problem:
 * Device or resource busy: it's mean your microphone is already used by another application
 * No such file or directory: it's mean no microphone found with your configuration

Notes: 
* You can only use one detector in MM
* Your mic can be used only by one application in MM

If you use your mic with another application it will not work (arecord limited command)

you can determinate your device with terminal command:
`arecord -l`
```sh
$ arecord -l
**** Liste des Périphériques Matériels CAPTURE ****
carte 2: Device [USB PnP Audio Device], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 0/1
  Sous-périphérique #0: subdevice #0
carte 3: Audio [Kinect USB Audio], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
```
select your prefered Card by using `device: "plughw:[card]"` and paste it in your configuration field
sample `device: "plughw:2"` if you want to use card nb 2

masoudl, 2020-08-07:
update: I actually got the detection to work by simply removing the settings for recorder and device, so the micconfig now is empty:
micConfig: { // put there configuration generated by auto-installer
},
...
New question: I see replies to all queries on the screen, but I don't get any audio playback. Is this module designed to also playback audio responses?

bugsounet, 2020-08-07:
1) if micConfig is empty, it's go back to default configuration. so you can delete it 
2) you have 2 solution for this (it's depend your configuration)
    * in RPI desktop, right click to volume icon and select your default input device (reboot after if needed)
    * If not worked again, try this: in `responseConfig: {}` you have `useNative` feature for more infomation read this [wiki part](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Configuration-Structure#field-responseconfig-)

masoudl, 2020-08-08:
Thank you. I was able to get youtube working beautifully.

closed 2020-08-08 by masoudl