bugsounet/MMM-GoogleAssistant #116: Blank screen after installation

Jouhney, 2020-08-07:
Hey everybody
I installed the module because I want a Google Assistant on my Mirror. But however, the screen is now just black after the installation. I think I did everything right. Does someone have an idea of what the issue is?

Thank you very much

bugsounet, 2020-08-07:
Hi,

if you have a "MODULE_NOT_FOUND" message, it's meems that all needed module was not installed correctly
maybe electron rebuild step ?
or maybe you have do `npm fix` command

in all case retry with `npm run rebuild` in GA directory and don't do another `npm` commands after installation done
If there is some error (red line)  when installing : paste the error part there (ALL red line)

Jouhney, 2020-08-07:
The last command helped. Thank you so much!

bugsounet, 2020-08-07:
perfect so ;)

closed 2020-08-07 by bugsounet