bugsounet/MMM-GoogleAssistant #120: Error during npm install

dhriti-krishna, 2020-08-10:
Hey,

while installing MMM-GoogleAssistant running **npm install** results in an error. Log below:
-----------------------------------------------------------------------------------------------------------------------
Welcome to MMM-GoogleAssistant

No root checking...
Checking MMM-GoogleAssistant: pi/pi
Checking MagicMirror: pi/pi

Checking OS...
OS Detected: linux-gnueabihf (raspbian 10 armv7l)
RPI pulseaudio check
pulseaudio is installed
Sorry, this module is not compatible with pulseaudio

npm ERR! code ELIFECYCLE
npm ERR! errno 255
npm ERR! MMM-GoogleAssistant@ preinstall: `installer/preinstall.sh`
npm ERR! Exit status 255
npm ERR! 
npm ERR! Failed at the MMM-GoogleAssistant@ preinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-08-10T04_18_01_520Z-debug.log
------------------------------------------------------------------------------------------------------
how do i solve this?

bugsounet, 2020-08-12:
I have just updated install script use pulseaudio
retry fresh install

dhriti-krishna, 2020-08-12:
Thanks a lot!

closed 2020-08-12 by dhriti-krishna