bugsounet/MMM-GoogleAssistant #121: jarvis seems locked

Saljoke, 2020-08-11:
Hello
If I don't use assistant for some time it happens that I can't activate it with voice command.
If I then activate it via Telegram it works correctly and seems to reactivate the voice command function.
It would appear as if the microphone is on standby.
Could it be a bug?
Anyway it rules!!

bugsounet, 2020-08-11:
Hi ;)

I learned about this bug by @sdetweil
I will take a look what happen really
maybe It's an `arecord` bug... just wait and see !
I will be back to you if I can found ;)

sdetweil, 2020-08-11:
@Saljoke   next time this happens, from a terminal window do 

ps - ef | grep arecord

does it return anything
like this  (good)
```
ps -ef | grep arecord
sam      31552 31507  0 09:00 ?        00:00:00 arecord -D plughw:2,0 -q -r 16000 -c 1 -t wav -f S16_LE -
sam      31637 11039  0 09:00 pts/39   00:00:00 grep --color=auto arecord
```
or only a line like this (bad)
```
ps -ef | grep arecord
sam      31328 11039  0 08:59 pts/39   00:00:00 grep --color=auto arecord
```

bugsounet, 2020-08-12:
Hi, i have just updated @bugsounet/snowboy (v1.5.1) and other sub-library

* can you try to update GA with `npm run update` command
* of course testing :) 
* and tell me if it's better.

I have ask to snowboy to restart automaticaly when 1Gb of data is received (eq env 8 hours of listening)
@sdetweil : maybe it's the same bug that we talked

sdetweil, 2020-08-12:
more than likely. our module checks and restarts at 1.5 gig.

I just had an error in the restart code. -r 16000 sends 4000 bytes each buffer. detected over limit, started restart routine, got ANOTHER  buffer, still over the limit, start restart routine AGAIN, oops. then disconnect mic..  

so now immediately reset counter, do can't do restart while already there

bugsounet, 2020-08-12:
I preferred secure to 1Gb, I'll see. anyway I would never find the problem because at home it is used once in the day

sdetweil, 2020-08-12:
no. silence counts too. never say a word, still fails

bugsounet, 2020-08-12:
in all case thx for the tips :)

Saljoke, 2020-08-12:
> @Saljoke next time this happens, from a terminal window do
>
> ps - ef | grep arecord
>
> does it return anything
> like this (good)
>
> ```
> ps -ef | grep arecord
> sam      31552 31507  0 09:00 ? [...]

Just a line....
Now I have to update and lets retry. I tell you next days.
Thx

bugsounet, 2020-08-12:
so, if you have just a line:  it's the bug of @sdetweil found
and **normaly** corrected in snowboy v1.5.1

Saljoke, 2020-08-14:
Solved!!!
Cool

closed 2020-08-14 by Saljoke