bugsounet/MMM-GoogleAssistant #122: Wrong aspect ratio on non HD Screen

PaulReichmuth, 2020-08-11:
Hey,

I am trying to get GA running on my Raspberry Pi Touchscreen, and its working wonderfully so far. My Problem is that the Overlay does not seem to be made for a 800x600 screen. I´ve messed around with the CSS a bit but couldn´t get any good results. Can you tell me what CSS class i need to use to reposition the Text and Logos correctly?

I´ve put a video of my problem down in the files section.
[VID_20200811_181426.zip](https://github.com/bugsounet/MMM-GoogleAssistant/files/5058126/VID_20200811_181426.zip)

Thanks!

PaulReichmuth, 2020-08-11:
I´ve figured it out. For those having the same problem, open `MMM-GoogleAssistant.css` and edit the `margin-top` value of the `#GA_TRANSCRIPTION p` Class

closed 2020-08-11 by PaulReichmuth

bugsounet, 2020-08-11:
in MMM-GoogleAssistant.css

GA_CONTENER -> popup translation bar
GA_STATUS-> logo status place
GA_STATUS.<status> -> status of GA
GA_TRANSCRIPTION -> live transcription
GA_LOGO -> GA logo