bugsounet/MMM-GoogleAssistant #124: audio check is failing during update

bravozulu1979, 2020-08-12:
Hi Bugsounet,

After several tries and installing MMM-GoogleAssistant module successfully, looks like I have broken some audio configuration, since the audio check during installation is failing and no micConfig{} is generated either. as a result, the GoogleAssistant Module in the config file is doing absolutely nothing. Can you please help ?

more details 👍 
at the prompt : Do you want check our audio aoutput configuration ..I hit Y and it says "Current Configuration not Working". Then it lists a selection of the microphone device .. card 0 sndrpigooglevoi .. when I enter card # as 0 and then it says " Make sure your microphone is on, press Yes and say something. I do that .. and then I see a message - 
"Using default output speaker for palying 
play WARN alsa : can't encode 0-bit unknown or not applicable
play WARN also : can't encode 16-bit Signed Integer PCM
testmic.wav

bugsounet, 2020-08-12:
humm ... it's seems that alsa is broken ...

* can you redefine your prefered output device in the RPI desktop ? (icon volume -> right click -> output -> select it)
* reboot your rpi
* retry `npm run update` for checking sound

bravozulu1979, 2020-08-12:
when I go to preferences -> Audio Device Settings , I get an error No ALSA audio devices were detected. Enable the internal audio device or connect a USD or HAT audio device. I assume HAT refers to the Google AIY kit ? I am reinstalling it the AIY kit right now

bravozulu1979, 2020-08-12:
latest update - AIY reinstall..did not change any thing .. now I am out of ideas. Please help!!

bugsounet, 2020-08-12:
sorry to tell you this but I don't have a Google AIY kit, so it's difficult too

bravozulu1979, 2020-08-12:
wait.. I am confused. So what is the voice input and output you use for the assistant module ? Is it possible to do without the Google AIY kit ?

lxne, 2020-08-12:
> wait.. I am confused. So what is the voice input and output you use for the assistant module ? Is it possible to do without the Google AIY kit ?

You do not need the Google AIY kit for this module to work. All you need is a raspberry and any kind of microphone that works with the pi (and some speakers or headphones if you wanna hear anything).

bugsounet, 2020-08-13:
Actually I have: 

* RPI4 (RPI3B+ works too) 
* [USB mic](https://www.amazon.fr/gp/product/B071171DBP/)
* [PIR Sensor](https://www.amazon.fr/gp/product/B01C6O9C38)
* [HP 2309v VGA monitor](https://www.lcd-compare.com/moniteur-HP2309V-HEWLETT-PACKARD-HP-2309v.htm)
* [HDMI -> VGA apdator](https://www.amazon.fr/gp/product/B01LXRVO8Z/)

I use main analogic 3.5 jack output for sound

closed 2020-09-08 by bugsounet