bugsounet/MMM-GoogleAssistant #125: Issue with registering device

Beastlynoob, 2020-08-12:
Hello, I am having trouble when I try to register the google assistant in my magic mirror to the app. I am on the 2nd to last step however I am having an issue. When I type  

```
curl -s -X POST -H "Content-Type: application/json" \
> -H "Authorization: Bearer $ACCESSTOKEN" -d @deviceInstance.json \
> https://embeddedassistant.googleapis.com/v1alpha2/projects/YOUR_PROJECT_ID/devices/
```

It gives me an error in terminal that says

```
{
  "error": {
    "code": 403,
    "message": "The request is missing a valid API key.",
    "status": "PERMISSION_DENIED"
  }
}
```

do you know why I am having this issue?

bugsounet, 2020-08-12:
Do you change url with your personal value of `YOUR_PROJECT_ID` ?

in curl POST url: embeddedassistant.googleapis.com/v1alpha2/projects/**YOUR_PROJECT_ID**/devices

Beastlynoob, 2020-08-12:
Yes I put in my own project Id and I still received the same issue.

Beastlynoob, 2020-08-28:
Do you know how to fix? I am very beggener so I might need help I also read wiki.

bugsounet, 2020-08-31:
> Do you know how to fix? I am very beggener so I might need help I also read wiki.

this part is for experimented user and for google gaction script making
if you don't use gaction, it's not needed

wiki says : `It's optional ! just for more control with gAction using and reserved for experimented user`

Note: gaction code is very very difficult and very difficult to apply to the mirror
needed: your OWN `js` and `json` code (and some google library) -> if you don't master this language ... it will be very very difficult
default result will be:
 * displayed an icon in Google Home App, and you can do nothing with it
 * waiting for coding information (with ssh command line)

In all case api key is not correct in your case  (it's write on the error message)

closed 2020-09-14 by bugsounet