bugsounet/MMM-GoogleAssistant #126: How can I connect this module to the google home app ?

Archimatt, 2020-08-12:
When I start my Magic mirror I see this screen ( without saying anything ) and I don't know where I can find the menu to sign in on my phone.
Do I have to connect it to the google home app ? and how ? 

[image]

Thanks for your answers 

( Ps : je suis français mais je ne sais pas si je peux écrire en fr ici )

bugsounet, 2020-08-13:
Hi,
I never had this kind of screen, weird
`brief Today` is a result of useBriefToday submodule of A2D but it seems do that screen. (try to disable it)

Maybe it's a problem with the token ?
can you try :
```
cd ~/MagicMirror/modules/MMM-GoogleAssistant
node auth_and_test.js
```
make a request (not in french, make your request in english)
and verify that is respond correctly

Tell me results

Archimatt, 2020-08-13:
Hey
I disabled useBriefToday so I don't have this problem anymore

I tried your command but the assistant doesn't answer, It's weird 
And do you know how to connect the Assistant to the Google Home app ?
[image]

bugsounet, 2020-08-20:
> And do you know how to connect the Assistant to the Google Home app ?

It don't have to ask you this.
I think you have make a mismake when you have created your `google setup`

fedej, 2020-08-25:
I was able to connect it to Google Home after following this wiki: https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Register-your-Mirror-as-Google-Assistant-related-device

Archimatt, 2020-08-26:
It worked for me, thanks !

closed 2020-08-26 by Archimatt