bugsounet/MMM-GoogleAssistant #127: Snowboy error

dhriti-krishna, 2020-08-13:
Hi, I have an issue with snowboy and I was hoping you could help me with it. 
Log below:

npm start

> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-08-13 17:16:22.907] [LOG]    Starting MagicMirror: v2.12.0
[2020-08-13 17:16:22.914] [LOG]    Loading config ...
[2020-08-13 17:16:22.919] [LOG]    Loading module helpers ...
[2020-08-13 17:16:22.920] [LOG]    No helper found for module: alert.
[2020-08-13 17:16:22.964] [LOG]    Initializing new module helper ...
[2020-08-13 17:16:22.965] [LOG]    Module helper loaded: updatenotification
[2020-08-13 17:16:22.965] [LOG]    No helper found for module: clock.
[2020-08-13 17:16:23.319] [LOG]    Initializing new module helper ...
[2020-08-13 17:16:23.320] [LOG]    Module helper loaded: calendar
[2020-08-13 17:16:23.320] [LOG]    No helper found for module: compliments.
[2020-08-13 17:16:23.321] [LOG]    No helper found for module: currentweather.
[2020-08-13 17:16:23.321] [LOG]    No helper found for module: weatherforecast.
[2020-08-13 17:16:23.352] [LOG]    Initializing new module helper ...
[2020-08-13 17:16:23.352] [LOG]    Module helper loaded: newsfeed
[2020-08-13 17:16:24.196] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v6.1-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-08-13 17:16:24.197] [LOG]    Loading module helpers ...
[2020-08-13 17:16:24.198] [LOG]    No helper found for module: alert.
[2020-08-13 17:16:24.198] [LOG]    Initializing new module helper ...
[2020-08-13 17:16:24.199] [LOG]    Module helper loaded: updatenotification
[2020-08-13 17:16:24.199] [LOG]    No helper found for module: clock.
[2020-08-13 17:16:24.200] [LOG]    Initializing new module helper ...
[2020-08-13 17:16:24.200] [LOG]    Module helper loaded: calendar
[2020-08-13 17:16:24.200] [LOG]    No helper found for module: compliments.
[2020-08-13 17:16:24.201] [LOG]    No helper found for module: currentweather.
[2020-08-13 17:16:24.201] [LOG]    No helper found for module: weatherforecast.
[2020-08-13 17:16:24.202] [LOG]    Initializing new module helper ...
[2020-08-13 17:16:24.202] [LOG]    Module helper loaded: newsfeed
[2020-08-13 17:16:24.207] [ERROR]  App threw an error during load
[2020-08-13 17:16:24.208] [ERROR]  Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v6.1-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js:8:29)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
[2020-08-13 17:16:24.209] [ERROR]  Whoops! There was an uncaught exception...
[2020-08-13 17:16:24.210] [ERROR]  Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v6.1-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js:8:29)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[2020-08-13 17:16:24.217] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-08-13 17:16:24.217] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-08-13 17:16:24.325] [LOG]    Launching application.
[2020-08-13 17:16:29.090] [LOG]    Shutting down server...
[2020-08-13 17:16:29.092] [LOG]    Stopping module helper: updatenotification
[2020-08-13 17:16:29.092] [LOG]    Stopping module helper: calendar
[2020-08-13 17:16:29.093] [LOG]    Stopping module helper: newsfeed
[2020-08-13 17:16:29.093] [LOG]    Stopping module helper: updatenotification
[2020-08-13 17:16:29.093] [LOG]    Stopping module helper: calendar
[2020-08-13 17:16:29.093] [LOG]    Stopping module helper: newsfeed
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Can you please help me with this?

bugsounet, 2020-08-13:
Hi,

Generally, there are 2 cases to have this errors:
 * You did not do the electron-rebuild step in `npm install`
 * You try to do `npm fix` or other npm command after installation was done

So retry with this command:
```
cd ~/MagicMirror/modules/MMM-GoogleAssistant
npm run rebuild
```
this command will erase all dependencies and do a clean reinstall of GA
and **Don't do other npm commands !** :)

Note:
If you have already determinate your micConfig, you can Ignore `Checking Speaker and Microphone...` step

dhriti-krishna, 2020-08-14:
Thanks a lot, Cédric. You really are very helpful. I'm going to close this case now. Thanks again!

closed 2020-08-14 by dhriti-krishna