bugsounet/MMM-GoogleAssistant #128: hotword detector (*.pmdl)

MrDondoT, 2020-08-19:
I've tried to use a personal model  (rachele.pmdl) as snowboy wake word in the config.js file

				`},
				snowboy: {
				  audioGain: 2.0,
				  Frontend: true, //use true with other models
				  Model: "rachele", 
				  Sensitivity: 0.7 //null
				},`

and put the file in the same dir of the other models:
/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/resources/models
but the mm will continue restarting with errors:

[image]

please advise
I also would suggest to write a note about it in the documentation.

bugsounet, 2020-08-20:
like i said on the forum i plan to do it ;)
Actually it's not coded in my npm library `@bugsounet/snowboy`  core

I will add it soon :)

bugsounet, 2020-08-23:
now coded with v2.4.2 (with @bugsounet/snowboy v1.6.0 npm library)

to update:
`npm run update` in the GA directory

and read wiki (snowboy part and structure)

bugsounet, 2020-08-23:
Note: I add a Global notice for @ktoanlba89 on start of @bugsounet/snowboy and GA/A2D : **copying my MagicMirror modules don't pay !**
I think, i will understand why...

MrDondoT, 2020-08-23:
@bugsounet I'll update later on tonight and test it.
thanks

closed 2020-08-25 by bugsounet