bugsounet/MMM-GoogleAssistant #130: jarvis immediately stops

WeridFire, 2020-08-24:
When i wake up jarvis with the hotword he starts and the default audio plays but then he immediately stops and says "Sorry but I don't understand.". I'm new to programming but i gave it a try and I'd like to complete this project please send help.

bugsounet, 2020-08-24:
euh, you have no programming to do (it's my job), for me programming means creating a module
so, you have just to configuring it ;)
Generaly, you have this error when you set an mismake `lang` in `assistantConfig: {}` part 
just read this [wiki part](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Configuration-Structure#field-assistantconfig-)

if it's correct for you, just past your config there, i will see

WeridFire, 2020-08-25:
Thanks for the quick reply, you were right there was a problem in "lang".

closed 2020-08-25 by WeridFire