bugsounet/MMM-GoogleAssistant #132: Regarding to Global Notice.

ktoanlba89, 2020-08-25:
Dear @bugsounet 

I'm really sorry about my fault on your package
I updated snowboy package here https://www.npmjs.com/package/@imtoanle/snowboy

Regarding to https://github.com/ktoanlba89/MMM-Assistant2Display.
I just forked it without author changes. And i'm using it together with MMM-GoogleAssistant for personal purpose (not commercial)

Again, Sorry you for any your frustrations.

bugsounet, 2020-08-25:
ah... just this npm package ...
copying **any** npm package, it's not a coding job. 
So play with me ... no problem ok ;)
in all case, any update and any bug can be correct in your case
So use it ... i will play too :+1:

part 1: for me is done (see my repo and npm source repo too , you will see EVERY NPM/GA/A2D npm sources is tagged )
and v1.6.0 of @bugsounet/snowboy is tagged too...
EVERYONE see this :
[2020-08-25 15:23:26.469] [LOG]    [SNOWBOY] snowboy v1.6.0 Initialized...
[2020-08-25 15:23:26.469] [LOG]    [@bugsounet] *************************
[2020-08-25 15:23:26.469] [LOG]    [@bugsounet] Special note to @ktoanlba89 (https://github.com/ktoanlba89)
[2020-08-25 15:23:26.469] [LOG]    [@bugsounet] Copying does not pay!
[2020-08-25 15:23:26.469] [LOG]    [@bugsounet] Try to make your own before delete my username !
[2020-08-25 15:23:26.469] [LOG]    [@bugsounet] Don't play with me guy !
[2020-08-25 15:23:26.469] [LOG]    [@bugsounet] *************************
very good publicity for you !!!

now I will play with part 2 (I let you discover in some days (maybe))

good luck to you, and good Fork (wow so many fork in your account... good programming !), and display all sources well :)
 * just change a name of module is not programming something
 * if I hide sources, it's not for nothing (but oupss... you prefer Fork and change name of the module)

ktoanlba89, 2020-08-25:
Dear @bugsounet 

I can't say anything about snowboy npm. That's total my fault. So nothing to say. just sorry you about it.

About A2D Module. Really I don't use Google Service for Assistant. I built another api instead Google SDK. That's reason i changed name of module. It can't call it as GoogleAssistant. :(

Can you tell me what i need to do ?

bugsounet, 2020-08-25:
it's weird, lpcm16 is now deleted too ...
Do not worry, it will be my turn to play,... and i LOVE this :) :)
In total transparency for all user... just an update and an magical script

no change to reduce it
bye

closed 2020-08-25 by bugsounet