bugsounet/MMM-GoogleAssistant #134: Plantage Magic Mirror

Icare75, 2020-08-25:
Bonjour,

le module MMM-GoogleAssistant a bien fonctionné après son installation. Mais depuis il me plante sans cesse MagicMirror au lancement. Une idée ?

bugsounet, 2020-08-25:
je suis dessus sur le probleme

Icare75, 2020-08-25:
Ca me rassure, je n'ai pas fait une mauvaise manip :-)

bugsounet, 2020-08-25:
c'est bon c'est réparer :)

Icare75, 2020-08-25:
Je débute sur RPI. Pour la mise à jour quelle est la commande ?

bugsounet, 2020-08-25:
npm run update

Icare75, 2020-08-25:
ça plante toujours pour moi..

bugsounet, 2020-08-25:
j'ai desoin de logs pour savoir

lancer le mirroir manuellement via 
```
cd ~/MagicMirror
npm start
```

Icare75, 2020-08-25:
c'est fait

Icare75, 2020-08-25:
A priori c'est le token qui a expiré
[2020-08-25 20:17:09.490] [LOG]    Starting MagicMirror: v2.12.0
[2020-08-25 20:17:09.497] [LOG]    Loading config ...
[2020-08-25 20:17:09.502] [LOG]    Loading module helpers ...
[2020-08-25 20:17:09.504] [LOG]    No helper found for module: alert.
[2020-08-25 20:17:09.555] [LOG]    Initializing new module helper ...
[2020-08-25 20:17:09.556] [LOG]    Module helper loaded: updatenotification
[2020-08-25 20:17:09.557] [LOG]    No helper found for module: clock.
[2020-08-25 20:17:09.926] [LOG]    Initializing new module helper ...
[2020-08-25 20:17:09.926] [LOG]    Module helper loaded: calendar
[2020-08-25 20:17:09.927] [LOG]    No helper found for module: compliments.
[2020-08-25 20:17:09.927] [LOG]    No helper found for module: currentweather.
[2020-08-25 20:17:09.928] [LOG]    No helper found for module: weatherforecast.
[2020-08-25 20:17:09.962] [LOG]    Initializing new module helper ...
[2020-08-25 20:17:09.962] [LOG]    Module helper loaded: newsfeed
[2020-08-25 20:17:10.242] [LOG]    Initializing new module helper ...
[2020-08-25 20:17:10.243] [LOG]    Module helper loaded: MMM-Traffic
[2020-08-25 20:17:11.184] [LOG]    Initializing new module helper ...
[2020-08-25 20:17:11.185] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-08-25 20:17:11.540] [LOG]    Initializing new module helper ...
[2020-08-25 20:17:11.541] [LOG]    Module helper loaded: MMM-NowPlayingOnSpotify
[2020-08-25 20:17:11.541] [LOG]    All module helpers loaded.
[2020-08-25 20:17:11.649] [LOG]    Starting server on port 8080 ... 
[2020-08-25 20:17:11.658] [LOG]    Server started ...
[2020-08-25 20:17:11.659] [LOG]    Connecting socket for: updatenotification
[2020-08-25 20:17:11.660] [LOG]    Connecting socket for: calendar
[2020-08-25 20:17:11.661] [LOG]    Starting node helper for: calendar
[2020-08-25 20:17:11.662] [LOG]    Connecting socket for: newsfeed
[2020-08-25 20:17:11.663] [LOG]    Starting node helper for: newsfeed
[2020-08-25 20:17:11.664] [LOG]    Connecting socket for: MMM-Traffic
[2020-08-25 20:17:11.665] [LOG]    MMM-Traffic helper started ...
[2020-08-25 20:17:11.665] [LOG]    Connecting socket for: MMM-GoogleAssistant
[2020-08-25 20:17:11.666] [LOG]    Connecting socket for: MMM-NowPlayingOnSpotify
[2020-08-25 20:17:11.666] [LOG]    Sockets connected & modules started ...
[2020-08-25 20:17:11.757] [LOG]    Launching application.
[2020-08-25 20:17:14.289] [LOG]    Create new calendar fetcher for url: http://calendar.google.com/calendar/ical/christophe.oxybol%40gmail.com/private-037d02f3f889fde0791310614e2c2e76/basic.ics - Interval: 300000
[2020-08-25 20:17:14.357] [LOG]    Create new news fetcher for url: https://www.francetvinfo.fr/titres.rss - Interval: 300000
[2020-08-25 20:17:14.674] [INFO]   Newsfeed-Fetcher: Broadcasting 31 items.
[2020-08-25 20:17:14.758] [ERROR]  (node:2706) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
[2020-08-25 20:17:14.762] [ERROR]  (node:2706) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
[2020-08-25 20:17:14.783] [INFO]   Checking git for module: MMM-Traffic
[2020-08-25 20:17:14.822] [INFO]   Checking git for module: MMM-GoogleAssistant
[2020-08-25 20:17:14.850] [INFO]   Checking git for module: MMM-NowPlayingOnSpotify
[2020-08-25 20:17:15.101] [LOG]    [ASSISTANT] MMM-GoogleAssistant Version: 2.4.3
[2020-08-25 20:17:15.223] [LOG]    Refreshed access token because it has expired. Expired at: 20:17:14 now is: 20:17:15
[2020-08-25 20:17:15.468] [ERROR]  [MD5] MMM-GoogleAssistant sources edited !
[2020-08-25 20:17:15.470] [ERROR]  [MD5] You don't have permission to do this !
[2020-08-25 20:17:15.471] [ERROR]  [MD5] Exiting MagicMirror...
pi@raspberrypi:~/MagicMirror $

bugsounet, 2020-08-25:
vous avez votre réponse :)

Icare75, 2020-08-25:
Juste une dernière question, comment on le réactive ?

bugsounet, 2020-08-25:
GA et A2D sont protégé (pour eviter le piratage de compte google par exemple, si qqn prendrai accès a votre RPI a distance)
 si  un de ses fichiers maitre est modifé, il ne se lancera pas

la meilleur piste pour revenir a la normal:
* sauvegarder votre credentials et token
* effacer complement le repertoire du module
* git clone le module et faire npm install (juste cette [étape](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#auto-install-for-raspberrypi-or-debian-linux-machines-only) d'auto install)
* pas desoin de retoucher au fichier de configuration config.js si il a deja été créé

Icare75, 2020-08-25:
Ok merci !!!

Icare75, 2020-08-25:
J'ai de nouveau mon token qui a expiré. J'ai l'impression que c'est lorsque je modifie le nom de jarvis dans  MMM-GoogleAssistant.js

closed 2020-08-25 by Icare75

bugsounet, 2020-08-26:
> J'ai de nouveau mon token qui a expiré. J'ai l'impression que c'est lorsque je modifie le nom de jarvis dans MMM-GoogleAssistant.js

`[2020-08-25 20:17:15.223] [LOG] Refreshed access token because it has expired. Expired at: 20:17:14 now is: 20:17:15`

ce n'est pas en rapport avec GoogleAssistant mais avec MMM-NowPlayingOnSpotify

tous mes logs sont Taggé (Soit le nom du module, soit le nom de la librarie)
[ASSISTANT], [AUDIO], [LPCM16], [SNOWBOY], etc...

Le programme de gestion des nom de l'assistant ne fait pas parti des librairies google. c'est un programme externe qui permet d'activer quelque chose (ici l'assistant)
sinon ce serait du `ok google` (mais pas dispo chez google sans payer des $$$)