bugsounet/MMM-GoogleAssistant #136: Unable to load the MMM-GoogleAssistant module

ericlim20, 2020-08-28:
When I run the magic mirror, I can't load the MMM-GoogleAssistant module with following error message:

[2020-08-28 12:41:27.994] [ERROR]  Error: Cannot find module './lib/node/index.js'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js

Can anyone help? Thanks in advance.

bugsounet, 2020-08-31:
can you try this: 

```
cd ~/MagicMirror/modules/MMM-GoogleAssistant
npm run update
```
and past me result (and your full process !) ?

ericlim20, 2020-09-01:
Hi bugsounet,

Thanks for your help.
There are some errors during update as below:

pi@raspberrypi:~ $ cd ~/MagicMirror/modules/MMM-GoogleAssistant
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ npm run update

> MMM-GoogleAssistant@2.4.6 update /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/update.sh

Welcome to GA updater !

Updating...
remote: Enumerating objects: 32, done.
remote: Counting objects: 100% (32/32), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 42 (delta 17), reused 18 (delta 16), pack-reused 10
Unpacking objects: 100% (42/42), done.
From https://github.com/bugsounet/MMM-GoogleAssistant
   112ada9..de4caa3  prod       -> origin/prod
   9f2cbc0..9aaefb2  dev        -> origin/dev
Updating 112ada9..de4caa3
Fast-forward
 MMM-GoogleAssistant.js     |   2 +-
 README.md                  |   1 +
 components/screenParser.js |   2 +-
 installer/postinstall.sh   | 106 +++++++++++-----------
 installer/preinstall.sh    |  19 +++-
 installer/update.sh        |  40 +++++++-
 package.json               |   4 +-
 recipes/with-status.js     |  21 +++++
 8 files changed, 129 insertions(+), 66 deletions(-)
 create mode 100644 recipes/with-status.js
Deleting ALL @bugsounet libraries...
Ready for Installing...

> MMM-GoogleAssistant@2.4.7 preinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/preinstall.sh

Welcome to MMM-GoogleAssistant "2.4.7",

Installing all npm libraries...

> @bugsounet/secure-ga@1.0.6 preinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/secure-ga
> node secure.js

[SECURE] Check MMM-GoogleAssistant digital footprint...
[SECURE] Ok, Good choice!
[SECURE] Let's continue installing...
npm ERR! code ENOENT
npm ERR! syscall chmod
npm ERR! path /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/secure-ga/node_modules/mkdirp/bin/cmd.js
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, chmod '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/secure-ga/node_modules/mkdirp/bin/cmd.js'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-09-01T16_22_32_367Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 254
npm ERR! MMM-GoogleAssistant@2.4.6 update: `installer/update.sh`
npm ERR! Exit status 254
npm ERR! 
npm ERR! Failed at the MMM-GoogleAssistant@2.4.6 update script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-09-01T16_22_32_591Z-debug.log

[2020-09-01T16_22_32_591Z-debug.log](https://github.com/bugsounet/MMM-GoogleAssistant/files/5157674/2020-09-01T16_22_32_591Z-debug.log)

bugsounet, 2020-09-01:
wired ... I don't use this

can you try `npm run rebuild` in GA directory ?

bugsounet, 2020-09-01:
I write it on [wiki](https://github.com/bugsounet/MMM-Assistant2Display/wiki/pir#how-do-some-simply-test-and-setup-sensibilitydelay-of-your-pir-sensor-)

marlonino, 2020-09-02:
Hy Bugsounet 
I had these problems during the update!

(env) pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ npm run update

> MMM-GoogleAssistant@2.4.7 update /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/update.sh

Welcome to GA updater !

Updating...
Already up to date.
installer/update.sh: linha 53: cd: /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules: Arquivo ou diretório inexistente
Deleting ALL @bugsounet libraries...
Ready for Installing...

> MMM-GoogleAssistant@2.4.7 preinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/preinstall.sh

Welcome to MMM-GoogleAssistant "2.4.7",

Installing all npm libraries...
npm WARN registry Unexpected warning for https://registry.npmjs.org/: Miscellaneous Warning EINTEGRITY: sha512-OuMub/6NU/0xWEzRz4t/Ldeuxq/PWJWNCez/9vVpeluN9cM/IL5/Xft5lXFDVbEAbp0k6kt1qoxg3ieqOXFDEg== integrity checksum failed when using sha512: wanted sha512-OuMub/6NU/0xWEzRz4t/Ldeuxq/PWJWNCez/9vVpeluN9cM/IL5/Xft5lXFDVbEAbp0k6kt1qoxg3ieqOXFDEg== but got sha512-z4PhNX7vuL3xVChQ1m2AB9Yg5AULVxXcg/SpIdNs6c5H0NE8XYXysP+DGNKHfuwvY7kxvUdBeoGlODJ6+SfaPg==. (0 bytes)
npm WARN registry Using stale data from https://registry.npmjs.org/ due to a request error during revalidation.
npm ERR! code EINTEGRITY
npm ERR! errno EINTEGRITY
npm ERR! Invalid response body while trying to fetch https://registry.npmjs.org/@bugsounet%2fsecure-ga: Integrity verification failed for sha512-OuMub/6NU/0xWEzRz4t/Ldeuxq/PWJWNCez/9vVpeluN9cM/IL5/Xft5lXFDVbEAbp0k6kt1qoxg3ieqOXFDEg== (/home/pi/.npm/_cacache/content-v2/sha512/3a/e3/2e6ffe8d53fd31584cd1cf8b7f2dd7aec6afcf58958d09ecfff6f5697a5b8df5c33f20be7f5dfb7995714355b1006e9d24ea4b75aa8c60de27aa39714312)

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-09-02T15_46_47_470Z-debug.log
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! MMM-GoogleAssistant@2.4.7 update: `installer/update.sh`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the MMM-GoogleAssistant@2.4.7 update script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm WARN Local package.json exists, but node_modules missing, did you mean to install?

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-09-02T15_46_47_612Z-debug.log

bugsounet, 2020-09-02:
it's means the sha512 sum of the downloaded file is not the same on the npm registry master server
(errors on download)

**maybe** by retry with `npm run rebuild` can help

ericlim20, 2020-09-02:
@bugsounet Thanks for your help. 
No more error after `npm run rebuild`. Really appreciate your help.

bugsounet, 2020-09-02:
I love my magical script ;)))

closed 2020-09-08 by bugsounet