bugsounet/MMM-GoogleAssistant #137: When MMM-GoogleAssistant update at lastest, There are some bugs.

angel20123, 2020-08-28:
When MMM-GoogleAssistant update at lastest, There are some bugs!!
(Before Updating, it work normally)

1. There are not find Library!
   - @bugsounet/npmcheck
   - @bugsounet/native-sound

2. After 2 Lib(above) installed manually, It boots normally. but When I say 'Smart Mirror' for working GoogleAssistant, The Programe is terminated with error(below)
  
  [MD5] ERROR [Error: ENOENT: no such file or directory, open 'modules/MMM-GoogleAssistant/components/sound.js'] {
  errno: -2,
  code: 'ENOENT',
  syscall: 'open',
  path: 'modules/MMM-GoogleAssistant/components/sound.js'
  }

bugsounet, 2020-08-28:
> When MMM-GoogleAssistant update at lastest, There are some bugs

hi, you post is so funny for me :)
there is no bugs ;)
So I will repond with a single sentence:

**Read Wiki**

wirdman03, 2020-08-28:
hey guys i have the same problem since my update but i maybe gonna try to reinstall the module

`pi@raspberrypi:~/MagicMirror $ npm run start

> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-08-28 14:01:04.828] [LOG]    Starting MagicMirror: v2.12.0
[2020-08-28 14:01:04.842] [LOG]    Loading config ...
[2020-08-28 14:01:04.854] [LOG]    Loading module helpers ...
[2020-08-28 14:01:04.859] [LOG]    No helper found for module: alert.
[2020-08-28 14:01:04.958] [LOG]    Initializing new module helper ...
[2020-08-28 14:01:04.961] [LOG]    Module helper loaded: updatenotification
[2020-08-28 14:01:04.964] [LOG]    No helper found for module: clock.
[2020-08-28 14:01:05.708] [LOG]    Initializing new module helper ...
[2020-08-28 14:01:05.710] [LOG]    Module helper loaded: calendar
[2020-08-28 14:01:05.713] [LOG]    No helper found for module: compliments.
[2020-08-28 14:01:05.715] [LOG]    No helper found for module: currentweather.
[2020-08-28 14:01:05.716] [LOG]    No helper found for module: weatherforecast.
[2020-08-28 14:01:05.732] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'grpc'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-08-28 14:01:05.734] [LOG]    Loading module helpers ...
[2020-08-28 14:01:05.735] [LOG]    No helper found for module: alert.
[2020-08-28 14:01:05.736] [LOG]    Initializing new module helper ...
[2020-08-28 14:01:05.737] [LOG]    Module helper loaded: updatenotification
[2020-08-28 14:01:05.738] [LOG]    No helper found for module: clock.
[2020-08-28 14:01:05.739] [LOG]    Initializing new module helper ...
[2020-08-28 14:01:05.740] [LOG]    Module helper loaded: calendar
[2020-08-28 14:01:05.741] [LOG]    No helper found for module: compliments.
[2020-08-28 14:01:05.742] [LOG]    No helper found for module: currentweather.
[2020-08-28 14:01:05.744] [LOG]    No helper found for module: weatherforecast.
[2020-08-28 14:01:05.753] [ERROR]  App threw an error during load
[2020-08-28 14:01:05.754] [ERROR]  Error: Cannot find module 'grpc'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js:5:14)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
[2020-08-28 14:01:05.756] [ERROR]  Whoops! There was an uncaught exception...
[2020-08-28 14:01:05.757] [ERROR]  Error: Cannot find module 'grpc'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/components/assistant.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
[... 3 lines cut ...]
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/google-assistant/components/assistant.js:5:14)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
[... cut ...]
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[2020-08-28 14:01:05.771] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-08-28 14:01:05.772] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-08-28 14:01:05.923] [LOG]    Launching application.

`

bugsounet, 2020-08-28:
**READ wiki for updating ...**
**I CAN'T READ IT FOR YOU !**

I will not respond of this kind of question !

Or, maybe just wiki is not needed, so i will delete it ??

wirdman03, 2020-08-28:
no sorry but i think it an another problem

bugsounet, 2020-08-28:
>  Error found: Error: Cannot find module 'grpc'

i don't use this library... since one mounth ago
so yes ... you don't read it

It's so really diffulcult to search `updating` word in the home page of the wiki ?

wirdman03, 2020-08-28:
no no sorry

bugsounet, 2020-08-28:
ok... so what command do you use ?

wirdman03, 2020-08-28:
yes that right i do a git pull but after i do npm update and now  it say when i start 

``

pi@raspberrypi:~/MagicMirror $ npm run start

> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-08-28 16:47:51.326] [LOG]    Starting MagicMirror: v2.12.0
[2020-08-28 16:47:51.340] [LOG]    Loading config ...
[2020-08-28 16:47:51.349] [LOG]    Loading module helpers ...
[2020-08-28 16:47:51.354] [LOG]    No helper found for module: alert.
[2020-08-28 16:47:51.443] [LOG]    Initializing new module helper ...
[2020-08-28 16:47:51.444] [LOG]    Module helper loaded: updatenotification
[2020-08-28 16:47:51.445] [LOG]    No helper found for module: clock.
[2020-08-28 16:47:52.143] [LOG]    Initializing new module helper ...
[2020-08-28 16:47:52.144] [LOG]    Module helper loaded: calendar
[2020-08-28 16:47:52.145] [LOG]    No helper found for module: compliments.
[2020-08-28 16:47:52.146] [LOG]    No helper found for module: currentweather.
[2020-08-28 16:47:52.147] [LOG]    No helper found for module: weatherforecast.
[2020-08-28 16:47:54.205] [LOG]    Initializing new module helper ...
[2020-08-28 16:47:54.207] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-08-28 16:47:54.210] [LOG]    No helper found for module: MMM-SwissCommute.
[2020-08-28 16:47:54.215] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: TypeError: Module is not a constructor
[2020-08-28 16:47:54.216] [LOG]    Loading module helpers ...
[2020-08-28 16:47:54.218] [LOG]    No helper found for module: alert.
[2020-08-28 16:47:54.219] [LOG]    Initializing new module helper ...
[2020-08-28 16:47:54.219] [LOG]    Module helper loaded: updatenotification
[2020-08-28 16:47:54.221] [LOG]    No helper found for module: clock.
[2020-08-28 16:47:54.222] [LOG]    Initializing new module helper ...
[2020-08-28 16:47:54.223] [LOG]    Module helper loaded: calendar
[2020-08-28 16:47:54.225] [LOG]    No helper found for module: compliments.
[2020-08-28 16:47:54.227] [LOG]    No helper found for module: currentweather.
[2020-08-28 16:47:54.228] [LOG]    No helper found for module: weatherforecast.
[2020-08-28 16:47:54.229] [LOG]    Initializing new module helper ...
[2020-08-28 16:47:54.230] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-08-28 16:47:54.231] [LOG]    No helper found for module: MMM-SwissCommute.
[2020-08-28 16:47:54.232] [ERROR]  App threw an error during load
[2020-08-28 16:47:54.234] [ERROR]  TypeError: Module is not a constructor
    at loadModule (/home/pi/MagicMirror/js/app.js:125:12)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
    at loadModule (/home/pi/MagicMirror/js/app.js:143:4)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
    at Class.loaded (/home/pi/MagicMirror/js/node_helper.js:18:3)
    at loadModule (/home/pi/MagicMirror/js/app.js:141:6)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
[2020-08-28 16:47:54.236] [ERROR]  Whoops! There was an uncaught exception...
[2020-08-28 16:47:54.237] [ERROR]  TypeError: Module is not a constructor
    at loadModule (/home/pi/MagicMirror/js/app.js:125:12)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
    at loadModule (/home/pi/MagicMirror/js/app.js:143:4)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
    at Class.loaded (/home/pi/MagicMirror/js/node_helper.js:18:3)
    at loadModule (/home/pi/MagicMirror/js/app.js:141:6)
    at loadNextModule (/home/pi/MagicMirror/js/app.js:158:5)
    at /home/pi/MagicMirror/js/app.js:160:6
[2020-08-28 16:47:54.245] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-08-28 16:47:54.246] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-08-28 16:47:54.382] [LOG]    Launching application.

``

bugsounet, 2020-08-28:
it's really a loool part this topic there ;)

bugsounet, 2020-08-28:
I says : 
> Ahh... i have done a git pull and it don't works ! But i don't read the wiki before
that the name of the issue !

I says: 
> READ wiki for updating ...

I says: 
> It's so really diffulcult to search updating word in the home page of the wiki ?

and you continue with git pull !

wirdman03, 2020-08-28:
no i already do now i try correct my problems

angel20123, 2020-08-29:
When I updated MMM-GoogleAssistant module, I searched wiki already. 

I used below commands.

-> cd ~/MagicMirror/modules/MMM-GoogleAssistant
-> npm run update

bugsounet, 2020-08-31:
can you retry and past me result ?

angel20123, 2020-09-01:
> can you retry and past me result ?

I tried to reinstall after MMM-GoogleAssistant removed. Everything is ok to working in raspberry. 

So.. I dont know now What is the problem..

thank you for  consideration!!

closed 2020-09-01 by angel20123