bugsounet/MMM-GoogleAssistant #141: MMM-GoogleAssistant Update Question

jmwyds, 2020-09-03:
Hey Cedric,

So I am updating the MMM-GoogleAssistant module in accordance with the instructions in the Wiki:
`npm run update`
However, when I run the command, I am getting npm errors and it causes my mirror to crash and open in its [Electron] state
[image]
. I don't know whether this is a problem with the update script or if this is a problem on my end. However, I have included a couple of pictures to demonstrate the issue more clearly. The first is the terminal output when running npm run update and the second is the log that it directs me to in the error statement.

Thanks for your time!
Jack

[image]
[image]

bugsounet, 2020-09-03:
Hi, I don't know why you have this error.

Can you try `npm run rebuild` in Google Assistant directory ?
thank for the feellback

jmwyds, 2020-09-03:
Hey Cedric. Hm, strange. When I ran `npm run rebuild` I received the npm error which I saw after a fresh clone and npm install. I have included a picture of the error and its corresponding log output below:
[image]
[image]

bugsounet, 2020-09-14:
please update to last node / npm version.

you are using old version

closed 2020-09-14 by bugsounet