bugsounet/MMM-GoogleAssistant #143: wrong answers from google assistant

xelamod, 2020-09-06:
Hello,
I've installed everything it it looks like everything is working, however I'm not getting the right answers from google assistant.
for example:
Q: what is the time?  A: "Here's what I found on the web..."
Q: what is the weather tomorrow? A: "Here's the top search results"
Q: how long to drive to work? A: "I'm affried I can't find directions to your destination"

what am I doing wrong?

thank you!

bugsounet, 2020-09-06:
unfortunately, it is the google server that gives the answers, I have no control over it.
I already have your case a few months ago ... it returned to normal a few days later.

Google Assistant SDK is +/- a demo of Google Assistant
Sometimes weird stuff comes out ...
I try to do the best with it

xelamod, 2020-09-06:
Thank you!

And one more question, is it possible to use a button (MMM-buttons) to stop a playing YouTube video for example?

thanks,
Alex Modelevich | IT Manager
96 Herbert St,<x-apple-data-detectors://0/1>
Northcote, VIC, 3070, Australia<x-apple-data-detectors://0/1>
☎ 03 9916 0704<tel:03%209916%200704> | 📱 0412 580 740<tel:0412%20580%20740> | 📠 +61 9916 0799<tel:+61%209916%200799>
✉ Alex.Modelevich@takingshape.com<mailto:Alex.Modelevich@takingshape.com> | wtakingshape.com<http://www.takingshape.com/>

bugsounet, 2020-09-06:
> And one more question, is it possible to use a button (MMM-buttons) to stop a playing YouTube video for example?

sorry, only coded without touch support 
only on vocal

closed 2020-09-08 by bugsounet