bugsounet/MMM-GoogleAssistant #151: Issue with Voice Recognition/Response to Keyword

SanketNayak2019, 2020-09-07:
I'm a new user and today I scrupulously installed the MMM-GA module but yet after adding the module to the config file and rebuilding electron, I am failing to get any response on the MagicMirror interface. I have checked that I am on the most recent version of MagicMirror and the GA module and I disabled the MMM-remote application. While updating the GA module, I checked my audio input and output configurations successfully. I was successfully able to run `node auth_and_test.js ` successfully and recieve personalized inputs from the Google server.

At this point I am unsure as to what could be causing my issue and seeing that I am not aware of a way to debug, I am blindly resting my suspicions on the keyword activation. I currently am using the alexa model and currently have my audio gain set to 5.0 and my sensitivity set to 1.0.

Below is the configuration I have for the GA module:
`{
		  module: "MMM-GoogleAssistant",
		  position: "fullscreen_above",
		  config: {
		    debug: false,
		    assistantConfig: {
		      lang: "en-US",
		      projectId: "", // Required to use gaction.
		      modelId: "", // (OPTIONAL for gaction)
		      instanceId: "", // (OPTIONAL for gaction)
		      latitude: 27.2038,
		      longitude: -77.5011,
		    },
		    responseConfig: {
		      useScreenOutput: true,
		      screenOutputCSS: "screen_output.css",
		      screenOutputTimer: 5000,
		      screenRotate: false,
		      activateDelay: 250,
		      useAudioOutput: true,
		      useChime: true,
		      newChime: false,
		      useNative: true,
		      playProgram: "mpg321"
		    },
		    micConfig: { // put there configuration generated by auto-installer
		      recorder: "arecord",
		      device: "plughw:2",
		    },
		    customActionConfig: {
		      autoMakeAction: false,
		      autoUpdateAction: false, // in RPI, gaction CLI might have some trouble.(current version should be 2.2.4, but for linux-arm, Google haven't updated) so leave this as false in RPI. I don't know it is solved or not.
		      actionLocale: "en-US", // At this moment, multi-languages are not supported, sorry. Someday I'll work.
		    },
		    snowboy: {
		      usePMDL: false,
		      audioGain: 5.0,
		      Frontend: true,
		      Model: "alexa",
		      Sensitivity: 1.0
		    },
		    A2DServer: {
		      useA2D: true,
		      stopCommand: "stop"
		    },
		    recipes: [ "with-MMM-TelegramBot.js" ],
		    NPMCheck: {
		      useChecker: true,
		      delay: 10 * 60 * 1000,
		      useAlert: true
		    }
		  }
		},
`

bugsounet, 2020-09-08:
Hi you can't use Alexa and MMM-GoogleAssistant at the same time (only one mic instence can be open)

SanketNayak2019, 2020-09-08:
When I referred to Alexa, I was referring to the snowboy implementation (`Model: "alexa"`) as referred to in the documentation.
[image]
Should I not be setting this value to alexa and if so what value should it be set to?

bugsounet, 2020-09-08:
sure `Model: "alexa"` is correct : 

adjust the device input (your mic) in the RapsberryPi desktop (or maybe to low or not enabled)
or the device `plughw` value is not correct ? (If not correct, i can't listen your mic)

see Log in the console, if any error it will be says to you

SanketNayak2019, 2020-09-08:
in-built pmdl files were not functioning in the environment. Solution was to download custom pmdl file from Snowboy

closed 2020-09-08 by SanketNayak2019