bugsounet/MMM-GoogleAssistant #155: Update notification for GoogleAssistant

tdungsi1, 2020-09-10:
Hello,
I updated GoogleAssistant to the newest version. I installed MMM-Assistant2Display and the npm check kept saying that "Update available for MMM-GoogleAssistant module, [NPM] @bugsounet/npmcheck v0.9.6 -> v0.9.7

[image]

bugsounet, 2020-09-10:
Hello,

precisely, I lost the wiki page ... you can remind me please?

tdungsi1, 2020-09-10:
Sorry, my bad.
May I ask you two more questions?
I tried to use personal hotword, I put the pmdl file into the components directory, then changed config.js model to The_Name
But It doesnt work. 

My config.js:
[image]

Name in the directory:
[image]

And my output on screen:

[image]

Thanks for helping.

bugsounet, 2020-09-10:
If you found the wiki page ... EVERYTHING is explain :)

I will not lost my time to copy/past !
sorry

bugsounet, 2020-09-10:
Wiki says : 
# Assume:
 * If you don't use this command, don't open ANY issue for updating problem... It's will be closed automaticaly without reponse !
 * If you don't want READ this part... so I don't want respond to your issue !

closed 2020-09-10 by bugsounet