bugsounet/MMM-GoogleAssistant #167: Updating GoogleAssistant kileld MagicMirror - are we due for an update or a new branch ?

arunfruity, 2020-09-21:
[2020-09-21 21:06:51.856] [ERROR]  [MD5] MMM-GoogleAssistant sources edited !
[2020-09-21 21:06:51.870] [ERROR]  [MD5] You don't have permission to do this
[2020-09-21 21:06:51.878] [ERROR]  [MD5] Maybe... You don't have permission to USE MMM-GoogleAssistant in MagicMirror! (thx to @MichM

I ran an update and ever since I am not able to run Magic Mirror with GoogleAssistant

fastastic, 2020-09-21:
Same for me... @bugsounet cmon dude don't be like that, you are acting like a kid

bugsounet, 2020-09-21:
hey guy... it's MY module.. i never asked you to use them. ok ?

@fastastic hey `kid` ... you don't like Read (but just write...) so try to read [this](https://github.com/bugsounet/MMM-GoogleAssistant#mmm-googleassistant) and [this](https://github.com/bugsounet/MMM-GoogleAssistant/issues/162) ?

bugsounet, 2020-09-21:
... and [There](https://github.com/bugsounet/MagicMirror) if you have a lot of courage `Kid` (sorry ... your picture is so funny for `Kid` !)

closed 2020-09-21 by bugsounet