bugsounet/MMM-GoogleAssistant #168: MMM-GoogleAssistant sources edited ! Help Please

mrandrian, 2020-09-23:
I had an error like this.. even though I never edited it

[2020-09-23 16:52:09.348] [ERROR]  [MD5] MMM-GoogleAssistant sources edited !
[2020-09-23 16:52:09.351] [ERROR]  [MD5] You don't have permission to do this
[2020-09-23 16:52:09.353] [ERROR]  [MD5] Maybe... You don't have permission to USE MMM-GoogleAssistant in MagicMirror! (thx to @MichMich)
[2020-09-23 16:52:09.355] [ERROR]  [MD5] Exiting MagicMirror...

timbuktu031, 2020-09-23:
Me too! 
Same Error after update.

zapman90, 2020-09-23:
Having the same issue as well! Never touched any of the files. Tried to re-install it a few times and also made sure to follow the guide properly. API seams to work properly during installation (Voice and Mic too). "MMM-GoogleAssistant sources edited" closes MagicMirror fast after startup.

Code from terminal;
`[2020-09-23 17:45:15.971] [ERROR]  [MD5] MMM-GoogleAssistant sources edited !
[2020-09-23 17:45:15.978] [ERROR]  [MD5] You don't have permission to do this
[2020-09-23 17:45:15.983] [ERROR]  [MD5] Maybe... You don't have permission to USE MMM-GoogleAssistant in MagicMirror! (thx to @MichMich)
[2020-09-23 17:45:15.986] [ERROR]  [MD5] Exiting MagicMirror...
`

pulmons4, 2020-09-23:
Hello guys, I have the same issue ...
Cedric, help us please !!!

tsxhyper, 2020-09-23:
Me too. I just downloaded everything and set it up. I'm really looking forward to using this module. Thanks for your hard work.

pulmons4, 2020-09-23:
I looked into Google Actions I wanted to check my credentials and I found this:
"Nos services sont actuellement indisponibles. Nous ne sommes pas en mesure de fournir des informations sur les quotas."
looks like there is an issue with Google Authentication ... maybe we just have to wait ?

bugsounet, 2020-09-23:
Euh... I don't know why I pin issue...

ahmedalkabir, 2020-09-24:
So, MMM-GoogleAssistant doesn't work for now ?? and if so when the update will come ???!!! 

I hope it's within this week @bugsounet because I really really count it for a graduate project.

bugsounet, 2020-09-24:
This week I think 🤔 ;)

tdh8316, 2020-09-25:
Hi, @bugsounet 

I'm planning to make an educational camp tomorrow, and participants will make their smart mirrors. Yesterday I rehearsed all courses and realized there was an error with the Google Assistant module. I spent a lot of time trying to find out why this error occurred then finally found this issue. At least, the notification at the top of the readme file would be helpful for users to know the cause of this error and that that is not their mistake.

I do not know correctly why this error occurs, but I know that this _broken-release_ can lead to the loss of reliability of many users to this module. I'm not sure whether or not it is possible because I have no clue why this error occurs at all, but the VCS may help to prevent this error.

I'm looking forward to I can use this module tomorrow. If your works not finished until then, I hope to know the solution to provide education to children.

I am sorry about the aggressive tone. But I would like tomorrow's camp goes well.

Thanks for this.

tdh8316, 2020-09-25:
PLEASE RESPOND, @bugsounet.

I MUST USE THIS MODULE WITH THE ORIGINAL MagicMirror SOFTWARE.
BECAUSE I HAVE ALREADY DONE TO WRITE MANUAL FOR TOMORROW'S CAMP.
PLEASE MAKE SURE IT IS POSSIBLE TO USE THIS MODULE WITH THE ORIGINAL MagicMirror SOFTWARE.

PLEASE, DO NOT SPOIL OUR CAMP.

-----------------------------------------
Okay. Finally, I understand what was happening between you and the [developer of MagicMirror](https://github.com/MichMich). I judged you have trouble with him, so didn't mention him.

Then, according to https://github.com/MichMich/MagicMirror/issues/2130, I assume that the conflict was solved.

https://github.com/bugsounet/MMM-GoogleAssistant/issues/167#issuecomment-696388897 - Kindly ask to you. Please reconsider your answer to @fastastic. Your statement that you never asked to use your modules was EXTREMELY RUDE. Well, I'm not sure if you know this, but **your cool modules probably have many users all over the world.** And because of the trouble between you and the MagicMirror developer, THESE A LOT OF **USERS ARE BEING HARMED**.

Just like you said https://github.com/MichMich/MagicMirror/issues/2129#issuecomment-695778920 if all the problems have been solved, that must to mean that your modules are works. But you know, currently, your modules are not available.

That's it. So, I'd like to ask you. Do you want people all over the world, who use your great module, are harmed?

**PLEASE REMEMBER THAT YOU ARE A BOSS DEVELOPER OF THIS MODULE.**
**AND I SINCERELY ASK FOR YOUR RESPONSIBLE BEHAVIOR.**

Hope you calm and resolve this issue.

bugsounet, 2020-09-25:
Hi, to all :) 

You know I'm doing my best at now!

The only concern is that I am very late ... (new job ... new accommodation -> move ...)

I'm going to do one thing NOW! because I really really are too late and it will be comlicated on my side!

I will therefore put ALL my modules online in order to make a smooth migration

But remember ... it's temporary :)

For disaccording, no one know what happen really... only Mich know what happen ... but he won't say anything in public (I understand it)

For the best ;)

@bugsounet

bugsounet, 2020-09-25:
Done GA is now open

try `npm run update` in the GoogleAssistant Directory for updating

timbuktu031, 2020-09-25:
Thank you. It's working great.

mrandrian, 2020-09-26:
Work again like a magic.

closed 2020-09-26 by mrandrian

tdh8316, 2020-09-26:
Thanks for your action, @bugsounet.
Thanks to your favor, today's camp successfully took place.

But I do not want occurring Galapagos syndrome of the MagicMirror.

Yesterday, I did know how to use this module even you had prohibited your modules from the original MagicMirror. By modifying the `md5/index.js` file, I could use this module.
Reference - I wrote [this](https://github.com/kanggo-sw/smart-mirror#%EA%B5%AC%EA%B8%80-%EC%96%B4%EC%8B%9C%EC%8A%A4%ED%84%B4%ED%8A%B8-%EB%AA%A8%EB%93%88-%EB%B3%80%EA%B2%BD%EC%82%AC%ED%95%AD).

But, I asked you to reconsider your behavior. That's because I was concerned about your fork falling behind.

I hope you to cooperate with the MagicMirror developer. The fight is only temporary, while the software is permanent.

Over the years, many tutorials using your Google Assistant module have written, whether or not you like them. But if you push ahead with replacing the MagicMirror, these tutorials and other's works will be useless.

In the spirit of the Open Source Community, I wish this conflict to resolve harmoniously.

Kind regards,

bugsounet, 2020-09-26:
that's why i have delete MD5 check and i will do other...

bugsounet, 2020-09-26:
@tdh8316 : for your camp, If you have any problem I'm there (or Telegram for live response)