bugsounet/MMM-GoogleAssistant #174: Jarvis doesn't want to wake up

Mike00001, 2020-09-30:
Hello @lxne @Saljoke  :smiley: 

I asked a question on the Magic Mirror forum and was advised to come and ask my question here.

I explain to you. Impossible to wake Jarvis.

After reading the wiki, I have carefully followed the installation instructions. Everything went very well, no error. The usb microphone records correctly in the console and it is audible in output (I even slightly increased the volume in alsamixer). The test with google assistant in console also works.

For code I just added the “simple” template in my config.js file and checked that the device number was ok.

After looking on the forum and all the issus on the project repo I really don't see where the problem is.

Additional Information: The microphone is only used for the google assistant

Do you have an idea? How can I help you so that you help me?

Have a good day :smile:

lxne, 2020-09-30:
So if you say "Jarvis" nothing is happening on the MagicMirror screen, right?

Can you post your config section and the output of 'arecord -l'.

Saljoke, 2020-10-01:
Hello
There is some errors on terminal when launching mirror?

Mike00001, 2020-10-06:
Hi, thanks for the reply :)

No nothing is happening when I call "Jarvis"

My config:
`{
                        module: "MMM-GoogleAssistant",
                        position: "fullscreen_above",
                        config: {
                                debug: false,
                                assistantConfig: {
                                        lang: "en-US",
                                        latitude: 50.85045,
                                        longitude: 4.34878,
                                },
                                micConfig: { // put there configuration generated by auto-installer
                                        recorder: "arecord",
                                        device: "plughw:2",
                                },
                        }
                        },
`
arecord -l output:
`**** Liste des Périphériques Matériels CAPTURE ****
carte 2: AK5371 [AK5371], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
`
good afternoon :)

Mike00001, 2020-10-12:
everything seems ok. Except for Local-Temperature but that's another problem.

main.js:478 Initializing MagicMirror.
translator.js:202 Loading core translation file: translations/fr.json
translator.js:227 Loading core translation fallback file: translations/en.json
loader.js:175 Load script: modules/default/alert//alert.js
module.js:486 Module registered: alert
loader.js:147 Bootstrapping module: alert
loader.js:175 Load script: modules/default/alert/notificationFx.js
loader.js:152 Scripts loaded for: alert
loader.js:194 Load stylesheet: modules/default/alert/notificationFx.css
loader.js:194 Load stylesheet: vendor/css/font-awesome.css
loader.js:154 Styles loaded for: alert
translator.js:175 alert - Load translation fallback: translations/en.json
/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:95 Electron Security Warning (Insecure Resources) This renderer process loads resources using insecure
  protocols.This exposes users of this app to unnecessary security risks.
  Consider loading the following resources over HTTPS or FTPS. 
 - http://localhost:8080/css/main.css
- http://localhost:8080/fonts/roboto.css
- http://localhost:8080/socket.io/socket.io.js
- http://localhost:8080/vendor/node_modules/nunjucks/browser/nunjucks.min.js
- http://localhost:8080/js/defaults.js
- http://localhost:8080/config/config.js
- http://localhost:8080/vendor/vendor.js
- http://localhost:8080/modules/default/defaultmodules.js
- http://localhost:8080/js/logger.js
- http://localhost:8080/translations/translations.js
- http://localhost:8080/js/translator.js
- http://localhost:8080/js/class.js
- http://localhost:8080/js/module.js
- http://localhost:8080/js/loader.js
- http://localhost:8080/js/socketclient.js
- http://localhost:8080/js/main.js
- http://localhost:8080/fonts/node_modules/roboto-fontface/fonts/roboto-condensed/Roboto-Condensed-Regular.woff2
- http://localhost:8080/translations/fr.json
- http://localhost:8080/translations/en.json
- http://localhost:8080/modules/default/alert//alert.js
- http://localhost:8080/modules/default/alert/notificationFx.js
- http://localhost:8080/modules/default/alert/notificationFx.css
- http://localhost:8080/vendor/css/font-awesome.css
- http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/all.min.css
- http://localhost:8080/vendor/node_modules/@fortawesome/fontawesome-free/css/v4-shims.min.css
  
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
warnAboutInsecureResources @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:95
/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:145 Electron Security Warning (Insecure Content-Security-Policy) This renderer process has either no Content Security
    Policy set or a policy with "unsafe-eval" enabled. This exposes users of
    this app to unnecessary security risks.
 
For more information and help, consult
https://electronjs.org/docs/tutorial/security.
 This warning will not show up
once the app is packaged.
(anonymous) @ /home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/renderer/security-warnings.js:145
loader.js:156 Translations loaded for: alert
loader.js:175 Load script: modules/default/updatenotification//updatenotification.js
module.js:486 Module registered: updatenotification
loader.js:147 Bootstrapping module: updatenotification
loader.js:152 Scripts loaded for: updatenotification
loader.js:154 Styles loaded for: updatenotification
loader.js:156 Translations loaded for: updatenotification
loader.js:175 Load script: modules/default/clock//clock.js
module.js:486 Module registered: clock
loader.js:147 Bootstrapping module: clock
loader.js:175 Load script: vendor/node_modules/moment/min/moment-with-locales.js
loader.js:175 Load script: vendor/node_modules/moment-timezone/builds/moment-timezone-with-data.js
[... 749 lines cut ...]
{"killed":false,"code":1,"signal":null,"cmd":"sudo modules/MMM-LocalTemperature/DHT 7 -m j -a 3","stderr":"Unable to read sensor data after 3 attempt(s).\n"}
log @ MMM-LocalTemperature.js:459
socketNotificationReceived @ MMM-LocalTemperature.js:245
(anonymous) @ module.js:243
(anonymous) @ socketclient.js:39
r.emit @ index.js:83
r.onevent @ index.js:83
MMSocket.self.socket.onevent @ socketclient.js:33
r.onpacket @ index.js:83
(anonymous) @ index.js:83
r.emit @ index.js:83
r.ondecoded @ index.js:83
(anonymous) @ index.js:83
r.emit @ index.js:83
a.add @ index.js:83
r.ondata @ index.js:83
(anonymous) @ index.js:83
r.emit @ index.js:83
r.onPacket @ index.js:83
(anonymous) @ index.js:83
r.emit @ index.js:83
r.onPacket @ index.js:83
r.onData @ index.js:83
ws.onmessage @ index.js:83
MMM-LocalTemperature.js:462 MMM-LocalTemperature: Request sent to get the sensor data.

bugsounet, 2020-10-12:
this kind of log will not talking...

Can you past the log with `npm start` please ?

```
bugs@debian:~$ cd MagicMirror/
bugs@debian:~/MagicMirror$ npm start

> magicmirror@2.13.0 start /home/bugs/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-10-12 17:25:03.131] [LOG]    Starting MagicMirror: v2.13.0
[2020-10-12 17:25:03.202] [LOG]    Loading config ...
[2020-10-12 17:25:03.236] [LOG]    Loading module helpers ...
[2020-10-12 17:25:03.254] [LOG]    No helper found for module: alert.
[2020-10-12 17:25:03.277] [LOG]    No helper found for module: clock.
...
```

we need this entire log ;)

Mike00001, 2020-10-13:
> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-10-13 09:22:22.308] [LOG]    Starting MagicMirror: v2.12.0
[2020-10-13 09:22:22.316] [LOG]    Loading config ...
[2020-10-13 09:22:22.320] [LOG]    Loading module helpers ...
[2020-10-13 09:22:22.322] [LOG]    No helper found for module: alert.
[2020-10-13 09:22:22.365] [LOG]    Initializing new module helper ...
[2020-10-13 09:22:22.365] [LOG]    Module helper loaded: updatenotification
[2020-10-13 09:22:22.366] [LOG]    No helper found for module: clock.
[2020-10-13 09:22:22.718] [LOG]    Initializing new module helper ...
[2020-10-13 09:22:22.718] [LOG]    Module helper loaded: calendar
[2020-10-13 09:22:22.719] [LOG]    No helper found for module: currentweather.
[2020-10-13 09:22:22.748] [LOG]    Initializing new module helper ...
[2020-10-13 09:22:22.748] [LOG]    Module helper loaded: newsfeed
[2020-10-13 09:22:22.772] [LOG]    Initializing new module helper ...
[2020-10-13 09:22:22.773] [LOG]    Module helper loaded: MMM-PIR-Sensor
[2020-10-13 09:22:22.775] [LOG]    Initializing new module helper ...
[2020-10-13 09:22:22.775] [LOG]    Check MagicMirror version for node helper 'MMM-LocalTemperature' - Minimum version:  2.2.1 - Current version: 2.12.0
[2020-10-13 09:22:22.776] [LOG]    Version is ok!
[2020-10-13 09:22:22.776] [LOG]    Module helper loaded: MMM-LocalTemperature
[2020-10-13 09:22:23.790] [LOG]    Initializing new module helper ...
[2020-10-13 09:22:23.790] [LOG]    Module helper loaded: MMM-Assistant2Display
[2020-10-13 09:22:26.111] [LOG]    Initializing new module helper ...
[2020-10-13 09:22:26.112] [LOG]    Module helper loaded: MMM-GoogleAssistant
[2020-10-13 09:22:26.112] [LOG]    All module helpers loaded.
[2020-10-13 09:22:26.216] [LOG]    Starting server on port 8080 ... 
[2020-10-13 09:22:26.225] [LOG]    Server started ...
[2020-10-13 09:22:26.226] [LOG]    Connecting socket for: updatenotification
[2020-10-13 09:22:26.227] [LOG]    Connecting socket for: calendar
[2020-10-13 09:22:26.227] [LOG]    Starting node helper for: calendar
[2020-10-13 09:22:26.228] [LOG]    Connecting socket for: newsfeed
[2020-10-13 09:22:26.228] [LOG]    Starting node helper for: newsfeed
[2020-10-13 09:22:26.229] [LOG]    Connecting socket for: MMM-PIR-Sensor
[2020-10-13 09:22:26.229] [LOG]    Connecting socket for: MMM-LocalTemperature
[2020-10-13 09:22:26.230] [LOG]    MMM-LocalTemperature: module loaded! Path: /home/pi/MagicMirror/modules/MMM-LocalTemperature
[2020-10-13 09:22:26.230] [LOG]    Connecting socket for: MMM-Assistant2Display
[2020-10-13 09:22:26.231] [LOG]    Connecting socket for: MMM-GoogleAssistant
[2020-10-13 09:22:26.231] [LOG]    Sockets connected & modules started ...
[2020-10-13 09:22:26.323] [LOG]    Launching application.
[2020-10-13 09:22:28.667] [LOG]    Create new calendar fetcher for url: http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics - Interval: 300000
[2020-10-13 09:22:28.690] [LOG]    Create new news fetcher for url: http://www.lesoir.be/rss/81853/cible_principale_gratuit?status=1 - Interval: 300000
[2020-10-13 09:22:29.169] [INFO]   Newsfeed-Fetcher: Broadcasting 15 items.
[2020-10-13 09:22:30.702] [INFO]   Calendar-Fetcher: Broadcasting 4 events.

lxne, 2020-10-16:
> * In config change 'debug' to 'true'

```
                       module: "MMM-GoogleAssistant",
                       position: "top_center",
                       config: {
                                debug: true,
                                ...
```
then ‘npm start'

ToniSkulj, 2020-10-24:
I had the same problem, for me there was a particular module (MMM-todoist) that somehow disallowed me of using MMM-GoogleAssistant, everything was fine but my MagicMirror did not respond to 'Jarvis'.

Try making a backup of your config.json, remove everything from the original except MMM-GoogleAssistant and see if works. If it does, try to figure out which module(s) is/are causing the problem.

Hope this helps!

bugsounet, 2020-11-21:
We think that you have solved your issue by your self, reopen it if needed

closed 2020-11-21 by bugsounet