bugsounet/MMM-GoogleAssistant #175: Unable to get to work

dyard12, 2020-10-01:
Hi, I followed all the instructions and without errors installing. I verified my mic is working. I just couldn't hear back during a test file.  I never see it wake up. So then I installed all the other modules as well thinking I might come across the issue. Then my screen just went black boot up. Now I have commented out the config portion for Google assistant. This is my 3rd time going back through each step to see if I can get this working. 

Thanks any help would be great.

Saljoke, 2020-10-01:
If the problem is GA you have to retry npm install or npm rebuild inside MMM-GoogleAssistant directory.
Maybe failed something. If you copy the log on terminal we can catch your problem...

dyard12, 2020-10-01:
thanks, I reinstall, can you show me a better example to add to the config when I look that the sample I'm not sure if I add that the way it is or modifies.
For example, do I use this simple or full - I checked my mic as well and see if record but I don't hear back, Can you tell me where the log is that you mentioned?  I searched but the logs I found have only one line
{
  module: "MMM-GoogleAssistant",
  position: "fullscreen_above",
  config: {
    assistantConfig: {
      latitude: 51.508530,
      longitude: -0.076132,
    },
    micConfig: { // put there configuration generated by auto-installer  -- I dont know what to put here
      recorder: "arecord",
      device: "plughw:2",
    },
  }
},
Thanks in advance help are much appreciated

Saljoke, 2020-10-01:
Try to delete this
recorder: "arecord",

And copy the terminal log 
Go to terminal and MagicMirror directory 
Start with npm start and ctrl m to see the terminal

dyard12, 2020-10-01:
This what I get when I add to the config

{
  module: "MMM-GoogleAssistant",
  position: "fullscreen_above",
  config: {
    assistantConfig: {
      latitude: 33.567684,
      longitude: -117.725609,
    },
    micConfig: { // put there configuration generated by auto-installer
      device: "plughw:2",
      recipes: "with-BackgroundStatus.js",
    },
  }
},

> magicmirror@2.12.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-10-01 11:55:37.083] [LOG]    Starting MagicMirror: v2.12.0
[2020-10-01 11:55:37.094] [LOG]    Loading config ...
[2020-10-01 11:55:37.105] [LOG]    Loading module helpers ...
[2020-10-01 11:55:37.109] [LOG]    No helper found for module: alert.
[2020-10-01 11:55:37.199] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:37.200] [LOG]    Module helper loaded: updatenotification
[2020-10-01 11:55:37.202] [LOG]    No helper found for module: clock.
[2020-10-01 11:55:37.875] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:37.876] [LOG]    Module helper loaded: calendar
[2020-10-01 11:55:37.877] [LOG]    No helper found for module: compliments.
[2020-10-01 11:55:38.440] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:38.441] [LOG]    Module helper loaded: MMM-NOAA3
[2020-10-01 11:55:38.481] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:38.482] [LOG]    Module helper loaded: newsfeed
[2020-10-01 11:55:38.485] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:38.486] [LOG]    Module helper loaded: MMM-MyScoreboard
[2020-10-01 11:55:39.993] [ERROR]  WARNING! Could not load config file.
Starting with default configuration. Error found: Error: Cannot find module
'./lib/node/index.js'
Require stack:
-
/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
-
/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
-
[2020-10-01 11:55:39.994] [LOG]    Loading module helpers ...
[2020-10-01 11:55:39.996] [LOG]    No helper found for module: alert.
[2020-10-01 11:55:39.997] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:39.998] [LOG]    Module helper loaded: updatenotification
[2020-10-01 11:55:39.999] [LOG]    No helper found for module: clock.
[2020-10-01 11:55:40.000] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:40.001] [LOG]    Module helper loaded: calendar
[2020-10-01 11:55:40.002] [LOG]    No helper found for module: compliments.
[2020-10-01 11:55:40.003] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:40.003] [LOG]    Module helper loaded: MMM-NOAA3
[2020-10-01 11:55:40.004] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:40.005] [LOG]    Module helper loaded: newsfeed
[2020-10-01 11:55:40.006] [LOG]    Initializing new module helper ...
[2020-10-01 11:55:40.006] [LOG]    Module helper loaded: MMM-MyScoreboard
[2020-10-01 11:55:40.012] [ERROR]  App threw an error during load
[2020-10-01 11:55:40.014] [ERROR]  Error: Cannot find module
'./lib/node/index.js'
Require stack:
-
/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
[... 36 lines cut ...]
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous>
(/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js:1:54)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [

'/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',

'/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[2020-10-01 11:55:40.030] [ERROR]  MagicMirror will not quit, but it might
be a good idea to check why this happened. Maybe no internet connection?
[2020-10-01 11:55:40.031] [ERROR]  If you think this really is an issue,
please open an issue on GitHub:
https://github.com/MichMich/MagicMirror/issues
[2020-10-01 11:55:40.191] [LOG]    Launching application.

Saljoke, 2020-10-01:
The problem is in installation
Try
cd modules/MMM-GoogleAssistant
git reset --hard
git pull
npm install

dyard12, 2020-10-01:
I did that added GA back to the config and started this is what's in the
logs. I get a black screen , I comment out and my magic mirror back to
normal

newest logs
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.8
3 info using node@v10.22.0
4 verbose stack Error: missing script: start
4 verbose stack     at run
(/usr/lib/node_modules/npm/lib/run-script.js:155:19)
4 verbose stack     at /usr/lib/node_modules/npm/lib/run-script.js:63:5
4 verbose stack     at
/usr/lib/node_modules/npm/node_modules/read-package-json/read-json.js:116:5
4 verbose stack     at
/usr/lib/node_modules/npm/node_modules/read-package-json/read-json.js:436:5
4 verbose stack     at checkBinReferences_
(/usr/lib/node_modules/npm/node_modules/read-package-json/read-json.js:391:45)
4 verbose stack     at final
(/usr/lib/node_modules/npm/node_modules/read-package-json/read-json.js:434:3)
4 verbose stack     at then
(/usr/lib/node_modules/npm/node_modules/read-package-json/read-json.js:161:5)
4 verbose stack     at
/usr/lib/node_modules/npm/node_modules/read-package-json/read-json.js:382:12
4 verbose stack     at
/usr/lib/node_modules/npm/node_modules/graceful-fs/graceful-fs.js:123:16
4 verbose stack     at FSReqWrap.readFileAfterClose [as oncomplete]
(internal/fs/read_file_context.js:53:3)
5 verbose cwd /home/pi/MagicMirror/modules/MMM-GoogleAssistant
6 verbose Linux 5.4.68-v7+
7 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
8 verbose node v10.22.0
9 verbose npm  v6.14.8
10 error missing script: start
11 verbose exit [ 1, true ]

Saljoke, 2020-10-01:
Are you sure MMM-GoogleAssistant is in modules directory?

dyard12, 2020-10-01:
yes
[image: image.png]

Saljoke, 2020-10-01:
Can't see your image

dyard12, 2020-10-01:
sorry about that
/home/pi/MagicMirror/modules/MMM-GoogleAssistant

dyard12, 2020-10-02:
I tried moving the position around I was thinking it might be that since
the screen is all black. Maybe my config entry is incorrect?
{
  module: "MMM-GoogleAssistant",
  position: "bottom_left",
  config: {
    assistantConfig: {
      latitude: "33.424.2616",
      longitude: "-1174318.948",
    },
   micConfig: { // put there configuration generated by auto-installer
      device: "plughw:2",
      recipes: "with-BackgroundStatus.js",
    },
  }
},

Saljoke, 2020-10-02:
No. But recipes is incorrect in mic config

bugsounet, 2020-10-05:
```
(/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js:1:54)
code: 'MODULE_NOT_FOUND',
```
you have not installed snowboy module correctly.
Please retry in GA directory:
```
npm run update
```
wiki says do electron rebuild step -> [so do it](https://github.com/bugsounet/MMM-GoogleAssistant/wiki/Installation#auto-install-for-raspberrypi-or-debian-linux-machines-only) !
wiki says don't do another command (npm fix, npm audit fix , ...)

thanks to read installation guide, configuration structure & sample

dyard12, 2020-10-08:
Thank I did the update and no more black screen. Now just does nothing wake
up.  Also, I have verified my mic works just troubleshooting why I
can't hear now

bugsounet, 2020-11-21:
We think that you have solved your issue by your self, reopen it if needed

closed 2020-11-21 by bugsounet