bugsounet/MMM-GoogleAssistant #176: Google assistant won't listen to what i say. It just says. I Sorry, i dont understand.

jorickjuh, 2020-10-08:
Google assistant won't listen to what i say. It just says. I Sorry, i dont understand. But i can active google with the hotword Ok google. Or Hey google.

Any ideas why? When i use npm start. it wont show any error. So idk

Saljoke, 2020-10-08:
Do you follow all wiki to activate the module? 
The config is right? We have to see some errors to have more information

mrandrian, 2020-10-19:
in my case it happened after I changed the latitude and longitude for my location in assistantConfig. But if I leave the value as the default config. GA is doing well. What should I do?

Saljoke, 2020-10-19:
We have to see your config if there is some errors and log on terminal

bugsounet, 2020-11-21:
We think that you have solved your issue by your self, reopen it if needed

closed 2020-11-21 by bugsounet