bugsounet/MMM-GoogleAssistant #177: Repeating itself

Surixx, 2020-10-15:
When i try to speak to the Assistant it is answering me but the same time i hear 2 times or more. And another issue is that speker problem sometimes its cracks and this time the restart help.

bugsounet, 2020-10-25:
I think you have launched GA 2 times ... (generally MMM-RemoteControl do it itself)

closed 2020-11-04 by bugsounet

Surixx, 2021-01-06:
And how can i fix this because i couldnt find any solution???

bugsounet, 2021-01-06:
in this case, you have 2 solutions

1) Don't use RemoteControl or eq
2) don't use module who can't be duplicate, sample:
  - Pir sensor module
  - Telegram module
  - Google Assistant Module
  - *** Google Modules (APIs)
  - Alexa module
  - Spotify module
  - Deezer module
  - *** Amazon modules