bugsounet/MMM-GoogleAssistant #179: Authentification problem (Oaut)

geoek, 2020-10-15:
Hi !
I have a problem during the authentification of application. I followed the tutorial instructions but here is the resultat on my miror : 

[image]

I try several settings on the OAut authorisation screen of google console, but it doesn't work (i think it is a bit different from when the tutorial was written.

Do you have an idea ? Thanks in advance

bugsounet, 2020-10-25:
yes it's possible, @sdetweil have report to me but not again correct it

Starzs906, 2020-11-25:
Im having the same problem. Is there any sort of fix ?

r4zi3lrrr, 2020-11-29:
I'm also getting the same!

bugsounet, 2020-12-08:
wiki upgraded, try sdetweil version of GA setup

closed 2020-12-28 by bugsounet