bugsounet/MMM-GoogleAssistant #180: Register your Mirror as Google Assistant related device

novak1510, 2020-10-17:
Hello, i followed step by step the install. But at the end, I have this type of message. 

(env) pi@raspberrypi:~ $ curl -s -X POST -H "Content-Type: application/json" \
> -H "Authorization: Bearer $ACCESSTOKEN" -d @deviceInstance.json \
> https://embeddedassistant.googleapis.com/v1alpha2/projects/mirroir-b8082/devices/
{
  "error": {
    "code": 404,
    "message": "Requested entity was not found.",
    "status": "NOT_FOUND"
  }
}

bugsounet, 2020-10-25:
maybe projectid is not correct ?

novak1510, 2020-10-27:
@bugsounet I checked my projectid & I tried several things. But it"s the same issue however i followed step by step the register. I don't understand

bumrocks, 2020-11-17:
I am having a very similar issue but with a slightly different error code...Upon using this code:

`ACCESSTOKEN=ya29.A0AfH6SMANI8RGn-56kGYfEfHGLbKAZpKK0JdA46WoNvabaciVinEEQub1qdrP3tgGn13ZlN18b5jgbecxvC9WHP_OQhHTuxmdTIj2pdcpvsZb6iSP0ni4acvDB6FsZhjIY1kMVCsIZ_XcTKfHO9urTVGUzDaeFWCiTpHjNCFLPPw`

and then:

`curl -s -X POST -H "Content-Type: application/json" \
-H "Authorization: Bearer $ACCESSTOKEN" -d @deviceInstance.json \
https://embeddedassistant.googleapis.com/v1alpha2/projects/magicmirror-75617/devices/`

I get:

`{
  "error": {
    "code": 400,
    "message": "Request contains an invalid argument.",
    "status": "INVALID_ARGUMENT"
  }
`

My brain is fried trying to decypher and figure everything out.  Sadly this "little" project was intended to be a birthday gift back in May!  Need help please!  I am not trying to hijack this thread  but hoping for some help for both since we are having this issue at the same point wihin the wiki.  Thank you in advance!

Beastlynoob, 2020-11-21:
I had a simmilar issue. Error 404 as well.

bugsounet, 2020-11-21:
please see Google tutorial (there](https://developers.google.com/assistant/sdk/reference/device-registration/register-device-manual)
It's +/- the same

so you have make an error somewhere or Google have missed to update it

In all case, if you don't code any GAction, this step is optional.

It's really complex to make MagicMirror and MMM-GoogleAssistant react with GAction

closed 2020-11-21 by bugsounet