bugsounet/MMM-GoogleAssistant #183: Fresh install - failure to generate a token

ivanovd, 2020-11-02:
MM ver. 2.13
RPi 3b+
Raspbian Buster

credentials.json file is present

Error during Google Assistant Setup:

`
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ node auth_and_test.js
/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/protobufjs/src/writer_buffer.js:1
���������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������

SyntaxError: Invalid or unexpected token
    at Module._compile (internal/modules/cjs/loader.js:723:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/protobufjs/src/index-minimal.js:14:25)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)

`

The installation was fresh and I see that the above error showed up there as well.

`
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ cd ..
pi@raspberrypi:~/MagicMirror/modules $ git clone https://github.com/bugsounet/MMM-GoogleAssistant
Cloning into 'MMM-GoogleAssistant'...
remote: Enumerating objects: 38, done.
remote: Counting objects: 100% (38/38), done.
remote: Compressing objects: 100% (31/31), done.
remote: Total 876 (delta 15), reused 16 (delta 7), pack-reused 838
Receiving objects: 100% (876/876), 2.86 MiB | 2.90 MiB/s, done.
Resolving deltas: 100% (426/426), done.
pi@raspberrypi:~/MagicMirror/modules $ cd MMM-GoogleAssistant
pi@raspberrypi:~/MagicMirror/modules/MMM-GoogleAssistant $ npm install

> MMM-GoogleAssistant@ preinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/preinstall.sh

Welcome to MMM-GoogleAssistant

No root checking...
Checking MMM-GoogleAssistant: pi/pi
Checking MagicMirror: pi/pi

Checking OS...
OS Detected: linux-gnueabihf (raspbian 10 armv7l)

Checking all dependencies...
All Dependencies needed are installed !

Checking GCC Version...
Do you want to check compatible GCC version [Y/n] 
Your choice: Y
GCC 7 is set by default

Installing @sdetweil sandbox fix...
fixing sandbox permissions

Installing Sources...
Branch 'prod' set up to track remote branch 'prod' from 'origin'.
Already up to date.

Installing all npm libraries...
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported

> @bugsounet/secure-ga@1.0.8 preinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/secure-ga
> node secure.js

[SECURE] Check MMM-GoogleAssistant digital footprint...
[SECURE] Ok, Good choice!
[SECURE] Let's continue installing...

> @bugsounet/snowboy@1.6.3 install /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy
> bash scripts/installer.sh

[@bugsounet/snowboy] Do you want to execute electron rebuild ? [Y/n]
Your choice: Y
/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/observable/bindCallback.js:1
������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������

SyntaxError: Invalid or unexpected token
    at Module._compile (internal/modules/cjs/loader.js:723:23)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/index.js:61:22)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)

> protobufjs@6.10.1 postinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/protobufjs
> node scripts/postinstall

> @bugsounet/snowboy@1.6.3 postinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy
> tsc -p tsconfig.json

> MMM-GoogleAssistant@2.4.13 postinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/postinstall.sh

Checking Speaker and Microphone...
Do you want check your audio output configuration [Y/n] 
Your choice: 
Checking audio input...
Make sure your microphone is on, press [Yes] and say something.
Press [No] if you don't want to check. [Y/n] 
Your choice: Y
Current configuration not Working !

Selection of the microphone device
**** List of CAPTURE Hardware Devices ****
card 2: CameraB404271 [USB Camera-B4.04.27.1], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
Indicate the card # to use [0-9]: 2
you have selected: plughw:2
Checking audio input...
Make sure your microphone is on, press [Yes] and say something.
Press [No] if you don't want to check. [Y/n] 
Your choice: Y
Using default output speaker for playing

testmic.wav:

 File Size: 96.0k     Bit Rate: 256k
  Encoding: Signed PCM    
  Channels: 1 @ 16-bit   
Samplerate: 16000Hz      
Replaygain: off         
  Duration: 00:00:03.00  

In:100%  00:00:03.00 [00:00:00.00] Out:48.0k [      |      ]        Clip:0    
Done.
Did you hear yourself? [Y/n] 
Your choice: Y
This is your GoogleAssistant micConfig working configuration :

micConfig: {
  recorder: "arecord",
  device: "plughw:2"
},

MMM-GoogleAssistant is now installed !
Press [Enter] to continue

npm notice created a lockfile as package-lock.json. You should commit this file.
added 308 packages from 225 contributors and audited 308 packages in 327.183s

8 packages are looking for funding
  run `npm fund` for details

found 1 high severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details
`

bugsounet, 2020-11-02:
I don't know why this library is corrupted.

I try fresh install and works.

can you try this: 

```
cd ~/MagicMirror/modules/MMM-GoogleAssistant
rm package-lock.json
rm -rf node_modules
```

this commands is eq to fresh install

ivanovd, 2020-11-02:
Thanks @bugsounet I will give it a try and report back.

ivanovd, 2020-11-02:
It did the trick. Thanks a lot!!!

closed 2020-11-03 by bugsounet