bugsounet/MMM-GoogleAssistant #186: Taille

Adel51, 2020-11-06:
Voici sur mon petit écran :

[image]

et voici sur le grand écran 

[image]

bugsounet, 2020-11-08:
bonjour, pour ceci je ne peux rien faire,

Je demande l'affichage de la page web en plein écran, donc c'est malheusement le site en lui meme qui demande ceci.
Donc c'est le site en lui meme qui n'est pas compatible avec votre résolution.

Essayez avec de faire un recherche sur wikipédia, je pense que cela devrai être correcte

Adel51, 2020-11-09:
Bonjour,
j'ai fait le teste avec une recherche Wiki et ça donne ceci : 
[image]

En fait, tout est petit sur ma TV (horloge, telegram, news, monitor, wifi, etc) : 
[image]

une idée?

Autre question : pensez vous que ce detecteur de mouvement pour arduino peut être utiliser pour le module PIR ?
voici l url du detecteur : 
https://www.bip-electronique.fr/cartes-de-developpement-et-accessoires-/4108-capteur-de-mouvement-pour-arduino.html

Merci d avance.

Adel51, 2020-11-14:
salut
j ai trouvé. J ai modifié un paramétrage sur la resolution dans buster ey c est beaucoup plus gros

closed 2020-11-14 by Adel51

bugsounet, 2020-11-14:
oui il suffit de zoomer