bugsounet/MMM-GoogleAssistant #187: Black screen on startup

Dantheman7188, 2020-11-08:
Hello, I followed the installation instructions however, when starting Magic Mirror it is stuck on a black screen that I can see my cursor on. If I comment out Google Assistant Magic mirror works normally.

This is what shows in my console after starting Magic Mirror:

pi@raspberrypi:~ $ cd MagicMirror
pi@raspberrypi:~/MagicMirror $ npm run start

> magicmirror@2.13.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2020-11-08 00:32:38.904] [LOG]    Starting MagicMirror: v2.13.0
[2020-11-08 00:32:38.912] [LOG]    Loading config ...
[2020-11-08 00:32:38.916] [LOG]    Loading module helpers ...
[2020-11-08 00:32:38.918] [LOG]    No helper found for module: alert.
[2020-11-08 00:32:38.968] [LOG]    Initializing new module helper ...
[2020-11-08 00:32:38.969] [LOG]    Module helper loaded: updatenotification
[2020-11-08 00:32:38.970] [LOG]    No helper found for module: clock.
[2020-11-08 00:32:39.374] [LOG]    Initializing new module helper ...
[2020-11-08 00:32:39.374] [LOG]    Module helper loaded: calendar
[2020-11-08 00:32:40.284] [ERROR]  WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v6.1-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[2020-11-08 00:32:40.285] [LOG]    Loading module helpers ...
[2020-11-08 00:32:40.286] [LOG]    No helper found for module: alert.
[2020-11-08 00:32:40.287] [LOG]    Initializing new module helper ...
[2020-11-08 00:32:40.287] [LOG]    Module helper loaded: updatenotification
[2020-11-08 00:32:40.288] [LOG]    No helper found for module: clock.
[2020-11-08 00:32:40.288] [LOG]    Initializing new module helper ...
[2020-11-08 00:32:40.288] [LOG]    Module helper loaded: calendar
[2020-11-08 00:32:40.294] [ERROR]  App threw an error during load
[2020-11-08 00:32:40.294] [ERROR]  Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v6.1-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js:8:29)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
[2020-11-08 00:32:40.296] [ERROR]  Whoops! There was an uncaught exception...
[2020-11-08 00:32:40.297] [ERROR]  Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v6.1-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js:8:29)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[2020-11-08 00:32:40.305] [ERROR]  MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[2020-11-08 00:32:40.305] [ERROR]  If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[2020-11-08 00:32:40.408] [LOG]    Launching application.

bugsounet, 2020-11-08:
Hi, electron rebuild missing for snowboy library

wiki says: 
```
Don't forget to do ALL electron-rebuild steps.
No other npm commands (e.g. npm audit fix) are needed after installation (they can break package dependencies)!
```

so retry installation with:

```
cd ~/MagicMirror/modules/MMM-GoogleAssistant
npm run rebuild
```

Dantheman7188, 2020-11-09:
It worked! Thank you very much!

closed 2020-11-09 by Dantheman7188