bugsounet/MMM-GoogleAssistant #188: no links are visually displayed to me

comturaRalph, 2020-11-08:
Hello,

First of all, many thanks for the great module!

I was able to install everything without errors.

Unfortunately, no links are visually displayed to me.

the entry in the config is "true"

Could you please help me?

many thanks

bugsounet, 2020-11-08:
Hi, there is many change in Google Server for SDK project, there is no many callback with links.

actually only weather and wikipedia works (maybe orther....)
When i have some time, i will see what can i do for Youtube, spotify and links features (maybe declared an app for open and discover the links)

comturaRalph, 2020-11-08:
Hello,

Thank you for your response.

For me, Wikipedia and the weather don't work either.

The weather is only visualized in the small display at the bottom of the screen.

bugsounet, 2020-11-08:
do you have set `electronOptions` ?

comturaRalph, 2020-11-09:
I think so. I added the following code in the config:

var config = {
  address: "localhost",
  electronOptions: {
    webPreferences: {
      webviewTag: true
    }
  },
  port: 8080,

do I have to do anything else?

comturaRalph, 2020-11-09:
it works :-)
However, I can't get Spotify to work.

I don't have a premium account, but it should still work, right?

My configuration is as follows:

    spotify: {
      useSpotify: true,
      useIntegred: true,
      useLibrespot: false,
      connectTo: "MMSpiegel",
      playDelay: 3000,
      minVolume: 10,
      maxVolume: 100,
      updateInterval: 1000,
      idleInterval: 10000,
      username: ".... @ googlemail.com",
      password: "xxxxxxxxxxx",
      CLIENT_ID: "c8 .......",
      CLIENT_SECRET: "94 ......",
      deviceDisplay: "Listening on",
      usePause: true,
      typeArtist: "artist",
      typePlaylist: "playlist",
      typeAlbum: "album",
      typeTrack: "track"
    },

I made the settings in the with-MMM-Spotify according to their instructions.

Do you have a tip why it still doesn't work?

comturaRalph, 2020-11-10:
Thank you for your help :-)

closed 2020-11-10 by comturaRalph