bugsounet/MMM-GoogleAssistant #191: I can't get Spotify to work.

comturaRalph, 2020-11-10:
Hallo,

I can't get Spotify to work. :-(

I don't have a Premium account, but it should still work, right?

My configuration is as follows:

spotify: {
  useSpotify: true,
  useIntegred: true,
  useLibrespot: false,
  connectTo: "MMSpiegel",
  playDelay: 3000,
  minVolume: 10,
  maxVolume: 100,
  updateInterval: 1000,
  idleInterval: 10000,
  username: ".... @ googlemail.com",
  password: "xxxxxxxxxxx",
  CLIENT_ID: "c8 .......",
  CLIENT_SECRET: "94 ......",
  deviceDisplay: "Listening on",
  usePause: true,
  typeArtist: "artist",
  typePlaylist: "playlist",
  typeAlbum: "album",
  typeTrack: "track"
},
I made the settings in the with-MMM-Spotify script according to the instructions.

I also wrote the recipe in the configuration. When called, the symbol (arrow and fireworks) appears in the top left as if it were starting, but unfortunately the music does not start.

Do you have a tip why it still doesn't work?

Many thanks in advance for your help

bugsounet, 2020-11-11:
Hi, can you read [this](https://github.com/bugsounet/MMM-Assistant2Display/wiki/spotify#spotify-free-members-restrictions) ?

comturaRalph, 2020-11-11:
oops ... I skipped that ...

Thank you anyway :-)

p.s. You have developed a great module here !!!

closed 2020-11-11 by comturaRalph