bugsounet/MMM-GoogleAssistant #195: node auth_and_test.js command giving an error message

T3SSERACT2, 2020-12-24:
Hi,
I have been trying to set up Google Assistant on my magic mirror and everything was going well until I arrived at this step: node auth_and_test.js . Whenever I put that in, I get this error message: 

`pi@raspberry:~/MagicMirror/modules/MMM-GoogleAssistant $ node auth_and_test.js

internal/modules/cjs/loader.js:638
    throw err;
    ^

Error: Cannot find module '@bugsounet/google-assistant'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:636:15)
    at Function.Module._load (internal/modules/cjs/loader.js:562:25)
    at Module.require (internal/modules/cjs/loader.js:692:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/auth_and_test.js:7:25)
    at Module._compile (internal/modules/cjs/loader.js:778:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:789:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
`
Does anyone have any advice for what to do here?
Thanks in advance.

bugsounet, 2020-12-24:
Hi, wiki says in step one: install the module
If you don't do it. You have this error (don't find the module)

T3SSERACT2, 2020-12-24:
Hi,
I've already installed the module,
I've followed all the steps from the Installation section and up until node auth_and_test.js in the Google Assistant Setup Section.

bugsounet, 2020-12-24:
Sure ;)

closed 2020-12-24 by bugsounet

T3SSERACT2, 2020-12-24:
ouch, I deleted and reinstalled the whole thing twice.