bugsounet/MMM-GoogleAssistant #198: I didn't hear back from GA

vtdang1194, 2020-12-28:
Thank you very much for providing great products.
I just followed wiki. But here is the result on my mirror
[image]
"mmap() failed: Không thể cấp phát bộ nhớ"
 (mmap() failed:cannot allocate memory)
This prevents GA from making a sound when responding

https://user-images.githubusercontent.com/75298030/103202349-bccaf500-4924-11eb-9644-068436218539.mp4

Do you have an idea ? Thanks in advance

bugsounet, 2020-12-28:
Hi,

yes I have an idea. Just read wiki ;)

closed 2020-12-28 by bugsounet

vtdang1194, 2020-12-28:
> Hi,
> 
> yes I have an idea. Just read wiki ;)

Yes! I have seen it! thank you :+1:

phungorquan, 2021-01-01:
Hi @vtdang1194  , could you open youtube or web when using G.A.
It could not work for since a month ago