bugsounet/MMM-GoogleAssistant #202: MagicMirror screen going blank/ error seen in logs

dhriti-krishna, 2021-01-06:
Hi, I have a problem with the Magic Mirror, whenever I run npm start. The screen goes blank and I see a lot of errors in the logs. The errors are as follows:-
-----------------------------------------------------------------------------------------------------------------------------
```
 WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module '@bugsounet/screen'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
```
```
[2021-01-06 21:05:00.869] [ERROR]  Error: Cannot find module '@bugsounet/screen'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js:1:240)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32)
[2021-01-06 21:05:00.872] [ERROR]  Whoops! There was an uncaught exception...
[2021-01-06 21:05:00.872] [ERROR]  Error: Cannot find module '@bugsounet/screen'
Require stack:
- /home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:627:15)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/electron/dist/resources/electron.asar/common/reset-search-paths.js:41:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Function.Module._load (internal/modules/cjs/loader.js:531:27)
    at Module.require (internal/modules/cjs/loader.js:685:19)
    at require (internal/modules/cjs/helpers.js:16:16)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js:1:240)
    at Module._compile (internal/modules/cjs/loader.js:786:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:798:10)
    at Module.load (internal/modules/cjs/loader.js:645:32) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-Assistant2Display/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
```

Any help would be great.
Thanks

bugsounet, 2021-01-06:
screen library is not installed, try to updade A2D

dhriti-krishna, 2021-01-07:
ok, i will try and let you know

dhriti-krishna, 2021-01-07:
Hi, i checked and i'm not able to update, since it resulted in an error
_____________________________________________________________________________________
```
pi@raspberrypi:~/MagicMirror/modules/MMM-Assistant2Display $ npm update

> @bugsounet/screen@1.1.2 preinstall /home/pi/MagicMirror/modules/MMM-Assistant2Display/node_modules/@bugsounet/screen
> sudo apt-get install cec-utils -y

Reading package lists... Done
Building dependency tree       
Reading state information... Done
cec-utils is already the newest version (4.0.4+dfsg1-2+rpi1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up aiy-python-wheels (1.4-1) ...
/var/lib/dpkg/info/aiy-python-wheels.postinst: line 9: pip3: command not found
dpkg: error processing package aiy-python-wheels (--configure):
 installed aiy-python-wheels package post-installation script subprocess returned error exit status 127
dpkg: dependency problems prevent configuration of aiy-bt-prov-server:
 aiy-bt-prov-server depends on aiy-python-wheels; however:
  Package aiy-python-wheels is not configured yet.

dpkg: error processing package aiy-bt-prov-server (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 aiy-python-wheels
 aiy-bt-prov-server
E: Sub-process /usr/bin/dpkg returned an error code (1)

> @bugsounet/secure-a2d@1.0.6 preinstall /home/pi/MagicMirror/modules/MMM-Assistant2Display/node_modules/@bugsounet/secure-a2d
> node secure.js

internal/bootstrap/switches/does_own_process_state.js:129
    cachedCwd = rawMethods.cwd();
                           ^

Error: ENOENT: no such file or directory, uv_cwd
    at process.wrappedCwd [as cwd] (internal/bootstrap/switches/does_own_process_state.js:129:28)
    at Object.resolve (path.js:978:47)
    at resolveMainPath (internal/modules/run_main.js:12:40)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:66:24)
    at internal/main/run_main_module.js:17:47 {
  errno: -2,
  code: 'ENOENT',
  syscall: 'uv_cwd'
}

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.1.2 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm ERR! code ELIFECYCLE
npm ERR! errno 100
npm ERR! @bugsounet/screen@1.1.2 preinstall: `sudo apt-get install cec-utils -y`
npm ERR! Exit status 100
npm ERR! 
npm ERR! Failed at the @bugsounet/screen@1.1.2 preinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-01-07T17_04_41_258Z-debug.log

```

I really hope you can help me with this

dhriti-krishna, 2021-01-07:
I tried to do npm install bugsounet/screen and i found an error there too.

`pi@raspberrypi:~/MagicMirror/modules/MMM-Assistant2Display $ npm install bugsounet/screen

> @bugsounet/screen@1.1.2 preinstall /home/pi/MagicMirror/modules/MMM-Assistant2Display/node_modules/@bugsounet/screen
> sudo apt-get install cec-utils -y

Reading package lists... Done
Building dependency tree       
Reading state information... Done
cec-utils is already the newest version (4.0.4+dfsg1-2+rpi1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up aiy-python-wheels (1.4-1) ...
/var/lib/dpkg/info/aiy-python-wheels.postinst: line 9: pip3: command not found
dpkg: error processing package aiy-python-wheels (--configure):
 installed aiy-python-wheels package post-installation script subprocess returned error exit status 127
dpkg: dependency problems prevent configuration of aiy-bt-prov-server:
 aiy-bt-prov-server depends on aiy-python-wheels; however:
  Package aiy-python-wheels is not configured yet.

dpkg: error processing package aiy-bt-prov-server (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 aiy-python-wheels
 aiy-bt-prov-server
E: Sub-process /usr/bin/dpkg returned an error code (1)
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.1.2 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.3: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

npm ERR! code ELIFECYCLE
npm ERR! errno 100
npm ERR! @bugsounet/screen@1.1.2 preinstall: `sudo apt-get install cec-utils -y`
npm ERR! Exit status 100
npm ERR! 
npm ERR! Failed at the @bugsounet/screen@1.1.2 preinstall script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2021-01-07T17_19_52_804Z-debug.log
`

bugsounet, 2021-01-07:
do you have see this ?
```
Setting up aiy-python-wheels (1.4-1) ...
/var/lib/dpkg/info/aiy-python-wheels.postinst: line 9: pip3: command not found
```
try to solve with install the missing command / package
weird i thought this package is installed by default on Buster (pip3)

closed 2021-01-07 by bugsounet

dhriti-krishna, 2021-01-08:
ok