bugsounet/MMM-GoogleAssistant #205: Electron rebuild failed

bruh02, 2021-01-13:
Hello

I'm running into the following error during auto-installation?

[@bugsounet/snowboy] Do you want to execute electron rebuild ? [Y/n]
Your choice: y
✖ Rebuild Failed

An unhandled error occurred inside electron-rebuild
make: Entering directory '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/build'
  CXX(target) Release/obj.target/snowboy/lib/snowboy.o
In file included from /home/pi/.electron-gyp/8.5.3/include/node/node.h:63,
                 from ../../../nan/nan.h:56,
                 from ../lib/snowboy.cc:1:
/home/pi/.electron-gyp/8.5.3/include/node/v8.h:8923:75: warning: ‘MicrotasksCompletedCallback’ is deprecated [-Wdeprecated-declarations]
   void AddMicrotasksCompletedCallback(MicrotasksCompletedCallback callback);
                                                                           ^
/home/pi/.electron-gyp/8.5.3/include/node/v8.h:8931:78: warning: ‘MicrotasksCompletedCallback’ is deprecated [-Wdeprecated-declarations]
   void RemoveMicrotasksCompletedCallback(MicrotasksCompletedCallback callback);
                                                                              ^
../lib/snowboy.cc: In constructor ‘SnowboyDetect::SnowboyDetect(const string&, const string&)’:
../lib/snowboy.cc:38:31: warning: catching polymorphic type ‘class std::runtime_error’ by value [-Wcatch-value=]
   } catch (std::runtime_error e) {
                               ^
In file included from ../../../nan/nan.h:56,
                 from ../lib/snowboy.cc:1:
../lib/snowboy.cc: At global scope:
/home/pi/.electron-gyp/8.5.3/include/node/node.h:566:43: warning: cast between incompatible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
       (node::addon_register_func) (regfunc),                          \
                                           ^
/home/pi/.electron-gyp/8.5.3/include/node/node.h:600:3: note: in expansion of macro ‘NODE_MODULE_X’
   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
   ^~~~~~~~~~~~~
../lib/snowboy.cc:183:1: note: in expansion of macro ‘NODE_MODULE’
 NODE_MODULE(SnowboyDetect, SnowboyDetect::Init)
 ^~~~~~~~~~~
In file included from /home/pi/.electron-gyp/8.5.3/include/node/node.h:63,
                 from ../../../nan/nan.h:56,
                 from ../lib/snowboy.cc:1:
/home/pi/.electron-gyp/8.5.3/include/node/v8.h: In instantiation of ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = node::ObjectWrap; T = v8::Object; typename v8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<node::ObjectWrap>&)]’:
/home/pi/.electron-gyp/8.5.3/include/node/node_object_wrap.h:84:78:   required from here
/home/pi/.electron-gyp/8.5.3/include/node/v8.h:10579:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfo<node::ObjectWrap>::Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<node::ObjectWrap>&)’} to ‘Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<void>&)’} [-Wcast-function-type]
                reinterpret_cast<Callback>(callback), type);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/.electron-gyp/8.5.3/include/node/v8.h: In instantiation of ‘void v8::PersistentBase<T>::SetWeak(P*, typename v8::WeakCallbackInfo<P>::Callback, v8::WeakCallbackType) [with P = Nan::ObjectWrap; T = v8::Object; typename v8::WeakCallbackInfo<P>::Callback = void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)]’:
../../../nan/nan_object_wrap.h:65:61:   required from here
/home/pi/.electron-gyp/8.5.3/include/node/v8.h:10579:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfo<Nan::ObjectWrap>::Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)’} to ‘Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<void>&)’} [-Wcast-function-type]
  SOLINK_MODULE(target) Release/obj.target/snowboy.node
/usr/bin/ld: cannot find -lcblas
collect2: error: ld returned 1 exit status
make: *** [snowboy.target.mk:145: Release/obj.target/snowboy.node] Error 1
make: Leaving directory '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:198:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.19.97-v7l+
gyp ERR! command "/usr/bin/node" "/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/.bin/node-gyp" "rebuild" "--target=8.5.3" "--arch=arm" "--dist-url=https://www.electronjs.org/headers" "--build-from-source" "--module_name=snowboy" "--module_path=/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v8.5-linux-arm" "--remote_path=./snowboy/v1.6.3/Release/" "--package_name=snowboy-v1.6.3-electron-v8.5-linux-arm.tar.gz" "--host=https://github.com/bugsounet/snowboy"
gyp ERR! cwd /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy
gyp ERR! node -v v10.23.1
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok 

Failed with exit code: 1

Error: make: Entering directory '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/build'
  CXX(target) Release/obj.target/snowboy/lib/snowboy.o
In file included from /home/pi/.electron-gyp/8.5.3/include/node/node.h:63,
                 from ../../../nan/nan.h:56,
[... 46 lines cut ...]
gyp ERR! cwd /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy
gyp ERR! node -v v10.23.1
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok 

Failed with exit code: 1
    at SafeSubscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/spawn-rx/lib/src/index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:156:26)
    at Subscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:92:26)
    at Subscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:72:18)
    at SafeSubscriber._next (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/spawn-rx/lib/src/index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:205:16)
    at SafeSubscriber.next (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/rxjs/internal/Subscriber.js:143:22)

> protobufjs@6.10.2 postinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/protobufjs
> node scripts/postinstall

> @bugsounet/snowboy@1.6.3 postinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy
> tsc -p tsconfig.json

> MMM-GoogleAssistant@2.4.15 postinstall /home/pi/MagicMirror/modules/MMM-GoogleAssistant
> installer/postinstall.sh

bugsounet, 2021-01-13:
> /usr/bin/ld: cannot find -lcblas

strange ... it should be will cover with `libatlas-base-dev` package and it will be forced to install with installer

retry with: 

```sh
sudo apt-get install libatlas-base-dev
```

and retry `npm install`

bruh02, 2021-01-14:
It worked

Small remark for future reference, <npm install> did not re-execute electron rebuild so I switched to

cd ~/MagicMirror/modules/MMM-GoogleAssistant
npm run rebuild

This seems to have done the trick, thanks a lot!

closed 2021-01-14 by bruh02