bugsounet/MMM-GoogleAssistant #207: Re Installing update

DJRamm, 2021-01-18:
Hi i have left a message about this earlier about me updating Google Assistant and then having just a black screen. I have a brain injury and sometimes things like this are easy and then there are time i want to kill mame and hurt very badly the person who puts this stuff together. I have read the wiki 3 to 4 times and i have found little things that i needed to do but still no joy. i'm at my wits end as i paid a tech on friday to fix up the lose end to get it running and now i've gone and fuck it.

Please Help

DJ Ramm

lxne, 2021-01-18:
Hi, there are two ways – I am aware of – to check if it is a Google Assistant issue.

1. Disable Google Assistant in the config.js by adding `disabled: true,` (e.g. below `module: "MMM-GoogleAssistant",`). Do the same for Assistant2Display if it is installed. Restart MM. If the black screen is gone the issue is definitely GA/A2D related.
2. Turn on debugging for the Google Assistant module `debug: true,` (e.g. below `config: {`). Start MM from the MM folder with `npm start`. Check the logs for GA issues.

Share your results here.

DJRamm, 2021-01-18:
How do i check the logs?

Thanks

DJ Ramm

lxne, 2021-01-18:
If you start MM with `npm start` you get a lot of lines in the terminal. These are the logs. Copy those and paste them here. And do not pay 160$ ever again to get MagicMirror to work. That is way too much.

DJRamm, 2021-01-18:
ok thank you

DJRamm, 2021-01-18:
npm ERR! code ENOENT
npm ERR! syscall open
npm ERR! path /home/pi/package.json
npm ERR! errno -2
npm ERR! enoent ENOENT: no such file or directory, open '/home/pi/package.json'
npm ERR! enoent This is related to npm not being able to find a file.
npm ERR! enoent 

@lxne 
this is the list

DJ Ramm

DJRamm, 2021-01-18:
this is the saved file. @lxne 
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.6
3 info using node@v10.22.1
4 verbose config Skipping project config: /home/pi/.npmrc. (matches userconfig)
5 verbose stack Error: ENOENT: no such file or directory, open '/home/pi/package.json'
6 verbose cwd /home/pi
7 verbose Linux 5.4.68-v7+
8 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
9 verbose node v10.22.1
10 verbose npm  v6.14.6
11 error code ENOENT
12 error syscall open
13 error path /home/pi/package.json
14 error errno -2
15 error enoent ENOENT: no such file or directory, open '/home/pi/package.json'
16 error enoent This is related to npm not being able to find a file.
17 verbose exit [ -2, true ]

DJ Ramm

lxne, 2021-01-18:
It looks as though you’re running npm start in the wrong directory. Please try `cd MagicMirror`
and then `npm start` in pi@raspberrypi:~/MagicMirror $

DJRamm, 2021-01-18:
ok

DJRamm, 2021-01-18:
@lxne here is the results from that, this time ended up with the black screen this time.

 magicmirror@2.14.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[18.01.2021 20:30.32.211] [LOG]   Starting MagicMirror: v2.14.0
[18.01.2021 20:30.32.221] [LOG]   Loading config ...
[18.01.2021 20:30.32.239] [LOG]   Loading module helpers ...
[18.01.2021 20:30.32.243] [LOG]   No helper found for module: alert.
[18.01.2021 20:30.32.476] [LOG]   Initializing new module helper ...
[18.01.2021 20:30.32.478] [LOG]   Module helper loaded: updatenotification
[18.01.2021 20:30.32.479] [LOG]   No helper found for module: clock.
[18.01.2021 20:30.34.033] [ERROR] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v8.5-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
[18.01.2021 20:30.34.036] [LOG]   Loading module helpers ...
[18.01.2021 20:30.34.038] [LOG]   No helper found for module: alert.
[18.01.2021 20:30.34.040] [LOG]   Initializing new module helper ...
[18.01.2021 20:30.34.041] [LOG]   Module helper loaded: updatenotification
[18.01.2021 20:30.34.042] [LOG]   No helper found for module: clock.
[18.01.2021 20:30.34.053] [ERROR] App threw an error during load
[18.01.2021 20:30.34.065] [ERROR] Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v8.5-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:798:15)
    at Function../lib/common/reset-search-paths.ts.Module._resolveFilename (electron/js2c/browser_init.js:7718:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:691:27)
    at Function.Module._load (electron/js2c/asar.js:748:26)
    at Module.require (internal/modules/cjs/loader.js:853:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js:8:29)
    at Module._compile (internal/modules/cjs/loader.js:968:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:986:10)
[18.01.2021 20:30.34.068] [ERROR] Whoops! There was an uncaught exception...
[18.01.2021 20:30.34.077] [ERROR] Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v8.5-linux-arm/snowboy.node'
Require stack:
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js
- /home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at Module._resolveFilename (internal/modules/cjs/loader.js:798:15)
    at Function../lib/common/reset-search-paths.ts.Module._resolveFilename (electron/js2c/browser_init.js:7718:16)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (internal/modules/cjs/loader.js:691:27)
    at Function.Module._load (electron/js2c/asar.js:748:26)
    at Module.require (internal/modules/cjs/loader.js:853:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js:8:29)
    at Module._compile (internal/modules/cjs/loader.js:968:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:986:10) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/index.js',
    '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[18.01.2021 20:30.34.082] [ERROR] MagicMirror will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[... cut ...]
[18.01.2021 20:30.34.420] [LOG]   Launching application.
[18.01.2021 20:30.34.451] [WARN]  (electron) The default value of app.allowRendererProcessReuse is deprecated, it is currently "false".  It will change to be "true" in Electron 9.  For more information please check https://github.com/electron/electron/issues/18397
[18.01.2021 20:30.46.718] [LOG]   Shutting down server...
[18.01.2021 20:30.46.725] [LOG]   Stopping module helper: updatenotification
[18.01.2021 20:30.46.727] [LOG]   Stopping module helper: updatenotification

Thanks

DJRamm

lxne, 2021-01-18:
There you can see the GoogleAssistant issues.
I would try...

```
cd ~/MagicMirror/modules/MMM-GoogleAssistant
npm run rebuild
```

DJRamm, 2021-01-18:
ok could you please point out the problems as even though i struggle with my memory due to my Brain injury i would like to try and do this myself and help others.

Thanking you

DJ Ramm

lxne, 2021-01-18:
This is the first appearance of the issue
> [18.01.2021 20:30.34.033] [ERROR] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v8.5-linux-arm/snowboy.node'

and there are more of the same down the way.

DJRamm, 2021-01-18:
ok i did read that and was thinking that was an important file. 

Thank you so much for you help @lxne 

DJ Ramm

DJRamm, 2021-01-18:
@lxne sorry but its not fully working my google calendar is now not showing up could please help or have i used up all my credits for today. LOL

DJ Ramm

bugsounet, 2021-01-18:
> There you can see the GoogleAssistant issues.
> I would try...
> 
> ```
> cd ~/MagicMirror/modules/MMM-GoogleAssistant
> npm run rebuild
> ```
> 
> wiki quote:
> > No other npm commands are needed after installation (it can brake dependencies) !
> > Electron-rebuild step is needed (it updates itself on its own in a short time)
> 

That why... I says Read wiki... I make many many hour to write it!
99% of response is on it

Install step... Update step... Rebuild step if mew magic mirror version 

After if you want many many command to write in terminal. I can delete all script. 
But i think really i will make so many time to install

DJRamm, 2021-01-18:
ok thank will do that 

DJ Ramm

DJRamm, 2021-01-18:
i have done that, the problem i'm having now is my Google calendar which was working before is now not. With the testing of the assistant is there a similar process for the calendar.

Thanking you

DJ Ramm

lxne, 2021-01-18:
I think your new issue is not related to GoogleAssistant.
Are you using the default calendar module?
If you start MM with `npm start` can you identify any issues in the log?

DJRamm, 2021-01-19:
I am using google calendar and yes i have tried npm start and there are no issues in the terminal but now i cannot remember the address of the folder where the log is saved.

Thanks

DJ Ramm

DJRamm, 2021-01-19:
@lxne i found it and this the last log.
0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.10
3 info using node@v10.23.1
4 verbose config Skipping project config: /home/pi/.npmrc. (matches userconfig)
5 verbose stack Error: ENOENT: no such file or directory, open '/home/pi/package.json'
6 verbose cwd /home/pi
7 verbose Linux 5.4.83-v7+
8 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
9 verbose node v10.23.1
10 verbose npm  v6.14.10
11 error code ENOENT
12 error syscall open
13 error path /home/pi/package.json
14 error errno -2
15 error enoent ENOENT: no such file or directory, open '/home/pi/package.json'
16 error enoent This is related to npm not being able to find a file.
17 verbose exit [ -2, true ]

Thanks

DJ Ramm

evgenblum, 2021-01-19:
I had the same problem after updating MagicMirror. First error - Error: Cannot find module '/home/pi/MagicMirror/modules/MMM-GoogleAssistant/node_modules/@bugsounet/snowboy/lib/node/binding/Release/electron-v8.5-linux-arm/snowboy.node'
It was solved by
cd ~/MagicMirror/modules/MMM-GoogleAssistant
npm run rebuild

But I have another problem when MMM-GoogleAssistant works with module MMM-Assistant2Display
The first error-
[19.01.2021 13:20.17.517] [ERROR] App threw an error during load
[19.01.2021 13:20.17.522] [ERROR] Error: /home/pi/MagicMirror/modules/MMM-Assistant2Display/node_modules/epoll/build/Release/epoll.node: file too short
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    
    Please, advise

bugsounet, 2024-06-30:
auto close old issues

closed 2024-06-30 by bugsounet