bugsounet/MMM-GoogleAssistant #493: GaxiosError

turncoat54, 2024-07-01:
Hi all,

I want to install MMM-GoogleAssistant just following the wiki.
But when I type the command `npm run token`. I opened the page which shows the code. But got `status: 405`. Am I wrong somewhere? Can anybody help me out? :)

```
~/workspace/MagicMirror/modules/MMM-GoogleAssistant $ npm run token

> MMM-GoogleAssistant@7.0.3 token
> installer/token.sh

Welcome to MMM-GoogleAssistant Token generator!

Note: The choice expects [Y or N] and does not need [Enter] to confirm
Do you want to install/reinstall MMM-GoogleAssistant token? [Y/n]
Your choice: y
Opening OAuth URL. Return here with your code.

If your browser will not open, you can copy/paste this URL:
 https://accounts.google.com/o/oauth2/v2/auth?access_type=offline&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype&prompt=consent&response_type=code&client_id=555974158735-tt5thpbq4hhr9fl5uftnr41764adrtcq.apps.googleusercontent.com&redirect_uri=https%3A%2F%2F
googleassistant.bugsounet.fr
Paste your code: 4/0ATx3LY4Qwe4u7ekDq71uODk6fQEJgfR7GXjdsZaacTFJnvd2s-OKjRJVTitvGQ2KPTu61A

Error getting tokens: GaxiosError
    at Gaxios._request (/home/zongru/workspace/MagicMirror/modules/MMM-GoogleAssistant/node_modules/gaxios/build/src/gaxios.js:142:23)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async OAuth2Client.getTokenAsync (/home/zongru/workspace/MagicMirror/modules/MMM-GoogleAssistant/node_modules/google-auth-library/build/src/auth/oauth2client.js:137:21) {
  config: {
    retry: true,
    retryConfig: {
      httpMethodsToRetry: [Array],
      currentRetryAttempt: 0,
      retry: 3,
      noResponseRetries: 2,
      statusCodesToRetry: [Array]
    },
    method: 'POST',
    url: 'https://oauth2.googleapis.com/token',
    data: '<<REDACTED> - See `errorRedactor` option in `gaxios` for configuration>.',
    headers: {
      'Content-Type': 'application/x-www-form-urlencoded',
      'User-Agent': 'google-api-nodejs-client/9.10.0',
      'x-goog-api-client': 'gl-node/20.12.2'
    },
    paramsSerializer: [Function: paramsSerializer],
    body: '<<REDACTED> - See `errorRedactor` option in `gaxios` for configuration>.',
    validateStatus: [Function: validateStatus],
    responseType: 'unknown',
    agent: HttpsProxyAgent {
      _events: [Object: null prototype],
      _eventsCount: 2,
      _maxListeners: undefined,
      options: [Object],
      requests: [Object: null prototype] {},
      sockets: [Object: null prototype] {},
      freeSockets: [Object: null prototype] {},
      keepAliveMsecs: 1000,
      keepAlive: false,
      maxSockets: Infinity,
      maxFreeSockets: 256,
      scheduling: 'lifo',
      maxTotalSockets: Infinity,
      totalSocketCount: 0,
      proxy: URL {},
      proxyHeaders: {},
      connectOpts: [Object],
      [Symbol(shapeMode)]: false,
      [Symbol(kCapture)]: false,
      [Symbol(AgentBaseInternalState)]: [Object]
    },
    errorRedactor: [Function: defaultErrorRedactor]
  },
  response: {
    config: {
      retry: true,
      retryConfig: [Object],
      method: 'POST',
      url: 'https://oauth2.googleapis.com/token',
      data: '<<REDACTED> - See `errorRedactor` option in `gaxios` for configuration>.',
      headers: [Object],
      paramsSerializer: [Function: paramsSerializer],
      body: '<<REDACTED> - See `errorRedactor` option in `gaxios` for configuration>.',
      validateStatus: [Function: validateStatus],
      responseType: 'unknown',
      agent: [HttpsProxyAgent],
      errorRedactor: [Function: defaultErrorRedactor]
    },
    data: '',
    headers: {
      allow: 'GET',
      'content-length': '0',
      date: 'Mon, 01 Jul 2024 12:24:16 GMT',
      vary: 'Origin'
    },
    status: 405,
    statusText: 'Method Not Allowed',
    request: { responseURL: 'https://oauth2.googleapis.com/token' }
  },
  error: undefined,
  status: 405,
  [Symbol(gaxios-gaxios-error)]: '6.3.0'
}
```

bugsounet, 2024-07-01:
Hi, I just Test with fresh credentials and result is: (with MMM-GoogleAssistant v7.1.0)

 `[GA] Token created!`

Don't forget to activate  `Google Assistant API` library
Don't Forget to configure `OAuth Censent screen`

Credentials type must be `OAuth client ID`
Application Type must be `Web application`

turncoat54, 2024-07-01:
Thank you @bugsounet for your quick reply.
I just found that there may be something wrong with my VPN network. I'll close this issue and check the network.
Thanks for your help!!!

closed 2024-07-01 by turncoat54