bugsounet/MMM-GoogleAssistant #494: Google Assistant not Activating with the "Jarvis" Command

emilegasbarri, 2024-07-02:
I have installed MMM-Google Assistant, EXT-Detector, EXT-Alert + EXT-Touch exactly per the wiki instructions.

When I tried to activate the Google Assistant with the Jarvis command, I received no response, yet the terminal showed " [DETECTOR] [RULES] Detected: jarvis from: Snowboy".

I also installed EXT-Touch to see if I could activate the assistant using my IR frame, but the button does not show up anywhere on MM despite installing it per the wiki instructions.

I have reinstalled MMM-GoogleAssistant and nothing changed.

Please help! Thank you.

bugsounet, 2024-07-02:
Hi, Maybe it's another module that crashed

Can you verify to disable all modules except:
 * clock
 * MMM-GoogleAssistant
 * EXT-Alert
 * EXT-Detector

Restart MM² and tell me if better

emilegasbarri, 2024-07-03:
Thanks for your response!

I've done so and I still get the same result. (response from [DETECTOR] in terminal but nothing on screen)

I've attached a screenshot of my config.js file contents which may help? 

[image]

turncoat54, 2024-07-03:
Is there any error log? Maybe something went wrong when executing the codes so that the interaction didn't show up.

emilegasbarri, 2024-07-04:
I don't believe there is any error log... I have "debug: true" for MMM-GoogleAssistant (as seen in the image in my last post).

I apologise as I am very new to this, I've pasted in what my terminal shows when starting MM, if that may help.

```
emilegasbarri@raspberrypi:~ $ cd MagicMirror
emilegasbarri@raspberrypi:~/MagicMirror $ npm run start

> magicmirror@2.27.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[2024-07-04 12:24:24.986] [LOG]   Starting MagicMirror: v2.27.0 
[2024-07-04 12:24:25.012] [LOG]   Loading config ... 
[2024-07-04 12:24:25.016] [LOG]   config template file not exists, no envsubst 
[2024-07-04 12:24:25.018] [LOG]   Loading module helpers ... 
[2024-07-04 12:24:25.027] [LOG]   Initializing new module helper ... 
[2024-07-04 12:24:25.028] [LOG]   Module helper loaded: MMM-GoogleAssistant 
[2024-07-04 12:24:25.029] [LOG]   No helper found for module: clock. 
[2024-07-04 12:24:25.031] [LOG]   Initializing new module helper ... 
[2024-07-04 12:24:25.031] [LOG]   Module helper loaded: EXT-Alert 
[2024-07-04 12:24:25.033] [LOG]   Initializing new module helper ... 
[2024-07-04 12:24:25.034] [LOG]   Module helper loaded: EXT-Detector 
[2024-07-04 12:24:25.034] [LOG]   All module helpers loaded. 
[2024-07-04 12:24:25.037] [LOG]   Starting server on port 8080 ...  
[2024-07-04 12:24:25.759] [LOG]   Server started ... 
[2024-07-04 12:24:25.760] [LOG]   Connecting socket for: MMM-GoogleAssistant 
[2024-07-04 12:24:25.760] [LOG]   Connecting socket for: EXT-Alert 
[2024-07-04 12:24:25.761] [LOG]   Starting module helper: EXT-Alert 
[2024-07-04 12:24:25.761] [LOG]   Connecting socket for: EXT-Detector 
[2024-07-04 12:24:25.761] [LOG]   Sockets connected & modules started ... 
[2024-07-04 12:24:25.979] [LOG]   Launching application. 
[4176:0704/122427.005805:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.006146:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.006483:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.006592:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.006766:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.006993:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.007348:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.007478:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.007722:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.007892:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.008156:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.008314:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.008635:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.008773:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.012169:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.012616:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.013260:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.013433:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.014582:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.016750:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.017052:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.017188:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.017428:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.017563:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.017799:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.017993:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[4176:0704/122427.018294:ERROR:gbm_wrapper.cc(76)] Failed to get fd for plane.: No such file or directory (2)
[4176:0704/122427.018440:ERROR:gbm_wrapper.cc(259)] Failed to export buffer to dma_buf: No such file or directory (2)
[2024-07-04 12:24:27.283] [INFO]  System information:
### SYSTEM:   manufacturer: ; model: ; raspberry: undefined; virtual: false
### OS:       platform: linux; distro: Debian GNU/Linux; release: 12; arch: arm64; kernel: 6.6.31+rpt-rpi-2712
### VERSIONS: electron: 29.1.6; used node: 20.9.0; installed node: 20.15.0; npm: 10.7.0; pm2: 5.3.1
### OTHER:    timeZone: Australia/Sydney; ELECTRON_ENABLE_GPU: undefined 
[2024-07-04 12:24:27.677] [LOG]   [GA] MMM-GoogleAssistant Version: 7.0.3 rev: 240614 
[... cut ...]
[2024-07-04 12:24:29.062] [LOG]   [DETECTOR] [LIBRARY] All libraries loaded! 
[2024-07-04 12:24:29.064] [LOG]   [DETECTOR] [DATA] Snowboy is initialized with 1 Models: jarvis 
[2024-07-04 12:24:29.064] [LOG]   [DETECTOR] [DATA] Initialized 
[2024-07-04 12:24:29.418] [LOG]   [DETECTOR] [RULES] Start listening. 1 Models 
[2024-07-04 12:24:34.140] [LOG]   [DETECTOR] [RULES] Stop listening. 
[2024-07-04 12:24:34.140] [LOG]   [DETECTOR] [RULES] Detected: jarvis from: Snowboy 
[4054:0704/122437.392723:ERROR:atom_cache.cc(229)] Add WM_CHANGE_STATE to kAtomsToCache"
```

turncoat54, 2024-07-04:
I'm also new to this. Bug I can show you my log. :)
```
[2024-07-04 21:37:41.749] [LOG]   [GA] You don't use PM2 
[2024-07-04 21:37:41.757] [LOG]   [GA] Assistant Ready! 
[2024-07-04 21:37:41.806] [INFO]  Checking git for module: EXT-Alert 
[2024-07-04 21:37:42.081] [INFO]  Calendar-Fetcher: Broadcasting 6 events from https://ics.calendarlabs.com/76/mm3137/US_Holidays.ics. 
[2024-07-04 21:37:42.107] [INFO]  Checking git for module: MagicMirror 
[2024-07-04 21:37:42.253] [LOG]   [DETECTOR] EXT-Detector Version: 2.7.0 rev: 240701 
[2024-07-04 21:37:42.264] [LOG]   [ALERT] EXT-Alert Version: 2.1.0 rev: 240701 
[2024-07-04 21:37:42.269] [LOG]   [DETECTOR] [LIBRARY] Ignored: ./components/porcupine 
[2024-07-04 21:37:42.825] [LOG]   [DETECTOR] [LIBRARY] Loaded: ./components/snowboy --> this.lib.Snowboy 
[2024-07-04 21:37:42.826] [LOG]   [DETECTOR] [LIBRARY] All libraries loaded! 
[2024-07-04 21:37:42.833] [LOG]   [DETECTOR] [DATA] Snowboy DetectorConfig: [ { detector: 'Snowboy', Model: 'jarvis', Sensitivity: null } ] 
[2024-07-04 21:37:42.837] [LOG]   [DETECTOR] [SNOWBOY] Checking models 
[2024-07-04 21:37:42.840] [LOG]   [DETECTOR] [SNOWBOY] Model selected: jarvis 
[2024-07-04 21:37:42.842] [LOG]   [DETECTOR] [SNOWBOY] Sensitivity set: 0.7,0.7 
[2024-07-04 21:37:42.848] [LOG]   [DETECTOR] [DATA] Snowboy is initialized with 1 Models: jarvis 
[2024-07-04 21:37:42.854] [LOG]   [DETECTOR] [DATA] Initialized 
[2024-07-04 21:37:43.378] [LOG]   [DETECTOR] [SNOWBOY] Starts listening. 
[2024-07-04 21:37:43.426] [LOG]   [DETECTOR] [LPCM16] Start listening: 1 channels - use: arecord - device: default - sample rate: 16000 
[2024-07-04 21:37:43.434] [LOG]   [DETECTOR] [RULES] Start listening. 1 Models 
[2024-07-04 21:37:55.808] [LOG]   [DETECTOR] [SNOWBOY] Detected: jarvis 
[2024-07-04 21:37:55.809] [LOG]   [DETECTOR] [SNOWBOY] Stops listening. 
[2024-07-04 21:37:55.811] [LOG]   [DETECTOR] [RULES] Stop listening. 
[2024-07-04 21:37:55.812] [LOG]   [DETECTOR] [RULES] Detected: jarvis from: Snowboy 
[2024-07-04 21:37:55.815] [LOG]   [DETECTOR] [LPCM16] Stop listening 
[2024-07-04 21:38:10.630] [LOG]   [DETECTOR] [SNOWBOY] Starts listening. 
[2024-07-04 21:38:10.681] [LOG]   [DETECTOR] [LPCM16] Start listening: 1 channels - use: arecord - device: default - sample rate: 16000 
[2024-07-04 21:38:10.685] [LOG]   [DETECTOR] [RULES] Start listening. 1 Models 
```

emilegasbarri, 2024-07-05:
Thank you @turncoat54 

Any chance you could upload a copy of your config.js file for me to compare with?

bugsounet, 2024-07-05:
@emilegasbarri : I have test your config and it's works

* Can you try to upgrade to MM² v2.28.0 (last version) ?
* Can you try to execute `npm run refresh` command inside `MMM-GoogleAssistant` folder ?

tell me if better ?
@bugsounet

emilegasbarri, 2024-07-06:
Thank you for your reply, again

I got it to work after reinstalling MMM-Detector and rewriting the config file.

closed 2024-07-06 by emilegasbarri

bugsounet, 2024-07-06:
perfect so,
Happy use