bugsounet/MMM-GoogleAssistant #507: [GA] [DATA] [ERROR] [FATAL] Assistant: tokenGA.json file not found !

MMAik100, 2024-08-06:
I had to rebuild my Rasberry PI4 and I am a beginner.   This is the latest error message.  Can someone advise me what to do?

lxne, 2024-08-06:
There obviously is no tokenGA.json in your GA-module folder. Do you have credentials?
https://wiki.bugsounet.fr/MMM-GoogleAssistant/SetupCredentials

MMAik100, 2024-08-07:
I did, but when I rebuilt the computer, I wiped everything instead of backing some stuff up.  I have the secret code and my google project is still there but I don't know how to generate or download the tokenGA again.

MMAik100, 2024-08-07:
i was able to figure it out.  thank you

bugsounet, 2024-08-07:
Hi, I think you forgot [this](https://wiki.bugsounet.fr/MMM-GoogleAssistant/Beforethefirststart#before-the-first-launch-of-mmm-googleassistant).

closed 2024-08-07 by lxne

MMAik100, 2024-08-09:
exactly.   I need to read more.  Thank you